zynthian/zynthian-webconf #5: RebootHandler.py syntax error

ulph, 2017-06-09:
10:27:15 zynthian zynthian_webconf.sh[298]: File "/zynthian/zynthian-webconf/lib/RebootHandler.py", line 17, in get
Jun 09 10:27:15 zynthian zynthian_webconf.sh[298]: check_output("(sleep 1 & reboot)&", shell=True)

forgot subprocess. ?

mheidt, 2017-06-09:
Could you elaborate?
I thought this is what the ampersand is doing...creating a new process in the background.
Is anything not working?
There is a timer, that is trying to reload the page...in Js. Maybe we need to change sth on that site.

ulph, 2017-06-10:
Sorry, I was in a rush. Title should be "webconf reboot option does not work", the rest is guessing after glancing log/code.

...

Problem was that webconf reboot option did not work as in system does not reboot. 

Checking the logs I was lead to believe the check_call line failed.

mheidt, 2017-06-10:
Yes, I can confirm. It just doesn't reboot.
from subprocess import check_output

is missing

Only tested the correct refresh webconf-behaviour after refactoring into different classes. (everything was in one single controller class before)
And locally testing, you don't want your dev-machine to reboot, so that I commented that line out :)

mheidt, 2017-06-10:
i cannot create a pull request for this feature only. we need to wait until Fernando has time to implement the ALSA Audio and Snaphot handling. Or he might change it in the master directly.

closed 2017-08-17 by mheidt