zynthian/zynthian-webconf #7: Add configuration for aubionotes

jofemodo, 2017-08-24:
Add a configuration line for aubionotes. It should allow to configure the command options for replacing in:

/etc/systemd/system/aubionotes.service

The new config option could be inside "Audio" settings, or create a new page, although i'm not sure of the  right title for that ;-)

mheidt, 2017-08-28:
What are the "command options"?
Or should we have something like the Wifi advanced config?
In this case a new "Aubionotes" menu item would be better?

jofemodo, 2017-08-28:
Ohhh, it's really simple. It's the command line options for the aubionotes command. A simple "text input field" should be enough. There are too many options. Perhaps adding a link to the aubionotes man page could help for people experimenting with this feature:

https://aubio.org/manpages/latest/aubionotes.1.html

mheidt, 2017-08-29:
So only the options of ExecStart, not the whole line . Not the /usr/local/bin/aubionotes

jofemodo, 2017-08-29:
Yes. But don't have to worry about replacing the string int the file:

/etc/systemd/system/aubionotes.service

I will do it in the sysem config script.

mheidt, 2017-08-29:
And you are going fill the ZYNTHIAN_AUBIONOTES_OPTIONS os.env during system start with the current value of the file?
Or do I have to read it from file, when the os.env is empty?

jofemodo, 2017-08-29:
Yes. I will. You can do exactly the same that is done for other config variables, like the "jackd options" in audio config.

mheidt, 2017-08-29:
checkout githubNo7.

I added the warning, if the system is going to be rebooted as well.

mheidt, 2017-09-04:
merged with #10

closed 2017-09-04 by mheidt