zynthian/zynthian-webconf #12: Backup / Restore

mheidt, 2017-09-04:
is it possible to add a kind of “backup/restore-my-sounds”? E.g. creating an archive (ZIP?) of

${HOME}/pedalboards
${ZYNTHIAN_SW_DIR}/mod-ui/dados/favorites.json
${ZYNTHIAN_MY_PLUGINS_DIR}/*
${ZYNTHIAN_MY_DATA_DIR}/*

jofemodo, 2017-09-20:
Hi @mheidt!
Is this task already finished?

mheidt, 2017-09-20:
If you did this?

This file is needed: /zynthian/config/backup_items.txt with the content:

${ZYNTHIAN_SW_DIR}/mod-ui/dados/favorites.json
${ZYNTHIAN_MY_PLUGINS_DIR}
${ZYNTHIAN_MY_DATA_DIR}

I did the work before you changed the config structure.
Maybe you should test it ;)

jofemodo, 2017-09-20:
Yes! It's in the zynthian-sys repository and created by udpate script.
I close the "Issue" ;-)

closed 2017-09-20 by jofemodo