zynthian/zynthian-webconf #15: Add webconf option to turn cursor in hdmi screen on/off

wyleu, 2017-09-18:
When using zynthian with an hdmi screen the lack of a cursor is an interface handicap.

Solution
Add an option in the display webconf option to turn this on and off.

mheidt, 2017-09-18:
self.request.arguments['DISABLE_CURSOR']

jofemodo, 2017-09-19:
Implemented in b94b697ed57ffd1f0fcf1c35c997fb9ee6526b0a

jofemodo, 2017-09-19:
Hi Markus!

I think this flag will fit better in the UI style section.
We don't need a full reboot after enabling/disabling the cursor.
Also, it should be "Enable Cursor", and its default value "False".
Sorry for not giving a good specification for the feature. I will do it better next time ;-)

jofemodo, 2017-09-20:
Move option from Display to UI style settings => 065c6aa417421c8e5aefc91d270e468b4efffd1f

closed 2017-09-20 by jofemodo