zynthian/zynthian-webconf #18: Add Audio option for RasPi built-in headphone jack

gmeader, 2017-09-19:
Please consider adding a new option on the Audio page: Built-in RasPi headphone jack. 

I know this will give bad audio quality, but it is useful for testing and playing with a new Zynthian when you have no audio device.

mheidt, 2017-09-20:
And what do you expect, the system does, when the option is checked? If you have an extra alsa volume, it is shown already.
Or do you mean, that the system works without additional sound card? 
So it's more like Built-In Audiocard?

gmeader, 2017-09-20:
I mean that the Zynthian system can be made to work without an audio card. 
Currently, there is no choice on the Audio configuration page for "No audio card" or "Use RaspberryPi headphone audio output", or "use HDMI audio output" (on the **form** where you **select** the audiocard you have installed)

mheidt, 2017-09-20:
I don't know, if this is even possible.
And I doubt if it makes sense. It will only be used by people, who want to know, if the system works.
But actually, when I started, I would liked to know, how it sounds. Talking about noise etc. 
What we need are sound samples to convince people to buy a decent soundcard for their raspberry.

mheidt, 2017-09-20:
If you find out, what has to be done in the scripts or running any commands, to make it work, we can build it. But I think, that Fernando and me won't have the time to dismantle a zynthian and check how a system without card works.

gmeader, 2017-09-20:
This is what I am trying to do:
http://wiki.zynthian.org/index.php/Simple_RPF_Touchscreen_Zynthian
(But without USB audio)
I think using no card just needs the correct Jackd settings, but I do not know what they are. I will try to figure that out. I once got mine working with no card, but when I updated, it stopped working, and I forgot what I did to make it work.

jofemodo, 2017-09-20:
Raspberry's native audio is really bad. The sound is very noisy and the latency, huge. You can't play anything with it. If it's included in the configuration, there should be a note explaining it. Anyway, it's not a priority task. If somebody (gmeader?) have the right parameters, we could add it ;-)

mheidt, 2017-11-16:
@gmeader 
Are you providing input or can we close this issue?

gmeader, 2017-11-16:
I'm mostly just documenting all the possibilities. It's up to Jofemodo.

jofemodo, 2017-11-17:
If nobody is giving a configuration guide for the RBPI's internal sound device, then we can close the issue ;-)

closed 2017-11-17 by jofemodo

softrabbit, 2019-06-25:
FYI, this is how I got it working on a Rpi3B running Gorgona. Under hardware/audio in the webconf:
- Soundcard: **Custom device**
- Config: **dtparam=audio=on**
- Jackd options: **-P 70 -t 2000 -s -d alsa -d hw:0 -r 44100 -P -p 1024 -n 2 -X raw**