zynthian/zynthian-webconf #19: Midi Filter/Translation

mheidt, 2017-10-01:
IGNORE PG
IGNORE CH#6:9 KP
IGNORE CH#6 CP
IGNORE PB
IGNORE CC#4
IGNORE CH#5:8 CC#16,17
TRANS CC#5 => CC#18
TRANS CH#4 CC#76 => CC#87
Where event types code are:

PG => Program change
KP => Polyphonic Key Pressure (Aftertouch)
CP => Channel Pressure (Aftertouch)
PB => Pitch Bending
CC => Continuous Controller Change

select Command [IGNORE|TRANS]
MutliSelect Channel [ALL (0), 1-16]
Select MidiEvent [PG…CC]
If MidiEvent is CC, the following will be displayed as well
MultiSelect CCValue [1…32?]
If command is TRANS the following is displayed:
SingleSelect TranslatedMidiEvent
SingleSelect TranslatedCCValue [1…32?]

jofemodo, 2017-10-04:
Hi Markus!

The parser is almost finished. The final location for the library is:

$ZYNTHIAN_UI_DIR/zyngine

For checking rules you can do like this:

```
import zynthian_midi_filter
try:
  MidiFilterRule("MAP CH#3 CC#5 => CH#4 CC#7")
except Exception e:
 print("ERROR parsing MIDI filter rule") 

```
Also, please, use "ZYNTHIAN_MASTER_MIDI_EVENT_MAP" as environment variable for the MIDI filter script ;-)

Regards,

mheidt, 2017-10-04:
File "./zynthian_webconf.py", line 43, in <module>
    from lib.MidiConfigHandler import MidiConfigHandler
  File "/home/markus/Documents/zynthian/zynthian-webconf/lib/MidiConfigHandler.py", line 10, in <module>
    import zynthian_midi_filter
  File "/zynthian/zynthian-ui/zyngine/zynthian_midi_filter.py", line 51, in <module>
    from zyncoder import *
ImportError: No module named 'zyncoder'

I had to make zyngine known with the following
import sys
sys.path.append('/zynthian/zynthian-ui/zyngine/')

jofemodo, 2017-10-05:
I'm using "map/mapping" instead of "translate", Could you change the code for having an coherent notation? ;-)

Thanks!!

mheidt, 2017-10-06:
Or we use both. Map/Mapping when it is generic (ignore or translate) and Translate, when it is the Non-Ignore Mapping.

mheidt, 2017-10-08:
I checked something in. But I couldn't make it work on my linux machine. Talking PYTHONPATH here.
And when I look into zyngine, I wonder how zynthian_controller should work with webconf.
Best would be if you integrate it and I could checkout master again before I do any more changes in an old branch.

jofemodo, 2017-10-09:
OK @mheidt! I will do it tomorrow.
Thanks! ;-)

jofemodo, 2017-10-10:
Hi @mheidt!

It's done. I've merge the last changes with githubNo19 branch and it's working more or less. 
The "composing rule" popup opens, but when you click to save, it doesn't work. It reloads the page and the rule is not added.  Also, if you reload the page, you should restore the "advanced check" status after, and things get more complex than needed. IMHO you should avoid "reloading" and use AJAX calls instead.  Perhaps you could compose the rule locally and check it using an AJAX call ;-)

Regards!

mheidt, 2017-10-11:
The following error is thrown, when I call the MidiFilterRule:

ERROR:root:ERROR parsing MIDI filter rule: name 'MidiFilterRule' is not defined

with 
IGNORE CH#1,2 PG  

then i changed the import to
from zyngine.zynthian_midi_filter import MidiFilterRule

And now it is
ERROR:root:ERROR parsing MIDI filter rule: 'NoneType' object has no attribute 'set_midi_filter_event_ignore'

I git pulled zyncoder and ui

jofemodo, 2017-10-11:
Hi @mheidt!

You have tu call the constructor with "False" as second parameter, so the library don't try to set the rule in the zyncoder engine and only perform the parsing (syntax check).

Regards!

mheidt, 2017-10-11:
You can test it.
It's not done yet.
The advancedCheckbox isn't ok. When you submit, advanced doesn't stay open.
I don't know how you want this unless I do it how i did it before.
And I guess, the parser needs more work as well.

jofemodo, 2017-10-11:
Hi @mheidt !

Don't worry about the config flag, i will save it as a session variable and restore when reloading ;-)

Regarding the parser ... what id the problem? For me it's working quite well ...

Regards,

mheidt, 2017-10-11:
@jofemodo 
I haven't had the time to look into it. 
But a \n seperated list of rules and one contains sth like
IGNORE CH#5:8 C_P_C#16,17

doesn't fail

jofemodo, 2017-10-11:
@mheidt 
For checking a list of \n-separated rules, you should use the "MidiFilterScript" class ;-)
MidiFilterRule is for checking one rule. Anyway, it should fail when passing a "list", so i will fix it.

Regards,

jofemodo, 2017-10-11:
OK! Fixed. Multi-line rules are not allowed anymore ;-)

mheidt, 2017-10-13:
@jofemodo 
check it out pls.

jofemodo, 2017-10-13:
OK Markus! I'm testing it ... ;-)

jofemodo, 2017-10-25:
Hi @mheidt !
I've been testing a little bit this feature and now it works quite well, although not perfect.

What i've been done:

- Improve the "advanced view" feature. Now a cookie is used for restoring the status after reloading the page.

- Fix the MIDI rule parser. In "MAP" commands, the number of elements of the channel list must be equal for arguments 1 and 2. If not channel list is specified, then all channels are used (1:16). When a channel list is specified for the first argument and no channel list is specified for the second one, then it will take the channel list from the first one. So now, you can write this rule:

`MAP CH#1 CC#2 => CC#1`

and it will work OK, as intuition suggests ;-)

In the other hand, there are some limitations that should be addressed:

1. First, note that MIDI protocol is a binary protocol and 0 is the first MIDI channel, aswell as 0 is the first CC number, etc. The Rule Parser use this convention. Of course, everybody is used to numbering the MIDI channels starting by 1, so i think the graphic tool should use this convention too, but generating the rules accordingly to the parser convention:

`Rule_Channel = GUI_Channel - 1`

2. The rule system allow cross-channel rules like that:

`MAP CH#1 CC#2 => CH#5 CC#4`

or

`MAP CH#1:4 CC#2 => CH#5:8 CC#4`

This should be allowed by the graphic tool, adding a second channel selector.

2. When selecting the CC number, only the first 32 controllers are allowed. It's very common to use higher values. IMHO the full range should be allowed (0-127), including CC#0 ;-)

And that's all by now !!

We are really close now ... come oooonnnnnnn!!! ;-)

jofemodo, 2017-10-25:
Perhaps the "rule builder" popup should be re-structured in 2 columns when building "MAP" rules?

mheidt, 2017-10-25:
ok, i will translate channel and cc value, but in the textarea, the translated value is shown.
We have the option of "telling the user" with a little info line.
Or your class class could provide a "translation" (again this word) method.
I prefer the former. Having two kind of rules stored in the system is error prone.

With the 2 column design, I don't hide/show the blocks, but disable them.
Maybe you want another css style for that.

jofemodo, 2017-10-26:
Regarding the 2-columns design for the popup, i like the current layout, but a different style (light grey?) should be used for "disabled" elements. You can use the "disable" selector in the css, so it's almost trivial to do.

A little detail: in firefox, when i add the first rule, an empty line is inserted first, giving a parser error. I will fix the parser to avoid the error. Anyway, it should be nice to remove this first "empty line".

Thanks!

mheidt, 2017-10-26:
And when you add the same line again, that is possible as well.
Your parser should have sth that eliminates them.
Esp. those that are semantically equal.

jofemodo, 2017-10-26:
You miss a space while generating the rules with CH in the second argument:

MAP CH#0 CC#3 => CH#0CC#6

;-)

jofemodo, 2017-10-26:
I've fixed the parser and now it will allow "empty lines" in the scripts. Also, it will allow to "comment" rules by adding "//" at begining.

mheidt, 2017-10-26:
fixed

jofemodo, 2017-10-27:
Hi @mheidt !

I just made some minor adjustments and merged the githubNo19 branch with master.

Congratulations for your work!  Now Zynthian can be used as a powerful MIDI filter. I'm testing it with one of the new Zynthian v2 with MIDI-OUT and it works like a charm!! Really!!

I would add more command, like channel transposing. Although this can be done from the UI, i think it can be useful to have these commands for using Zynthian as a standalone MIDI filter.

Thanks!!

jofemodo, 2017-10-27:
I've added some "commonly used" CC names and fixed the offset (CC#0 is Bank Select, CC#1 is Modulation). Take a look to this:

http://nickfever.com/music/midi-cc-list

Regards,

closed 2017-10-30 by jofemodo