zynthian/zynthian-webconf #20: Advanced View

mheidt, 2017-10-02:
Enable advanced view mechanism in order to hide expert fields

jofemodo, 2017-10-04:
Hi Markus!

I'm testing this feature and i found some problems. When checking/unchecking the option, page reloads and depending on the page, it restart the UI or reboot the Zynthian.

I think it would be a lot easier and faster if page is not reloaded. I always use "jquery" for this kind of things ... and you get some nice effects "for free" ;-)
Currently jquery is included, so you don't need to do nothing, only call "slideDown(500)" and "slideUp(500) with the right selector.

Regards!

mheidt, 2017-10-04:
I fixed this bug with my last commit. But I can change it to jquery-only later.

jofemodo, 2017-10-04:
Checkout this branch: gh20-advanced_options ;-)

mheidt, 2017-10-04:
forget my branch. I repeated some refactoring changes into the githubNo19 branch, but didn't pushed yet

PR #22 (2017-10-05): Gh20 advanced options
Implements Advanced Options checkbox (show/hide) using only javascript (JQuery).
Close #20.
commit d93b1240 (2017-10-05): Merge pull request #22 from zynthian/gh20-advanced_options

Gh20 advanced options. Close #20.
closed 2017-10-05 by jofemodo