zynthian/zynthian-webconf #23: Library Manager: Add a module for ZynAddSubFX banks & presets

jofemodo, 2017-10-25:
- User preset's dir for ZynAddSubFX => $ZYNTHIAN_MY_DATA_DIR/zynbanks
- Banks: Directory names. No special format. Bank numbers are assigned sequentially.
- Presets: XXXX-Preset_name.xiz, where XXXX is the program number with leading 0s.

It would be nice to have "Bank" numbers in the same way that we have Program numbers, but currently ZynAddSubFX works in that way ...

mheidt, 2017-10-25:
do you think, zynbanks is ok? isn't it zasfxbanks?
imagine another synthesizer within zynthian....

mheidt, 2017-10-25:
I thought, I would copy the snapshot page and enhance it with an upload of soundfonts?

Why does zasfx use 4 digits? program numbers are only 0-127? no?
Should I change the snapshot program nummer to 0 based numbering as well?

mheidt, 2017-10-27:
And we need the musical artifacts download

mheidt, 2017-10-28:
I go for zynbanks and then subdirectories for each file type. Like xiz!?

mheidt, 2017-10-29:
Or is it presets?

jofemodo, 2017-10-29:
It should be the same structure that the zynbanks directory under zynthian-data. ZynAddSubFX requires this structure:

* zynbanks
  * bank1
    * 0001-preset01.xiz
    * 0002-preset02.xiz
    * ...
  * bank2
    * 0001-preset01.xiz
    * 0002-preset02.xiz
    * ...
  * ...

The 4 digits is because ZASFX only use LSB for banks (127 banks) and use the bank MSB + preset number for allowing 65536 presets/bank. In fact it should be 5 digits ;-)

Regards,

mheidt, 2017-10-29:
Isn't it possible to change it?
There will be more requirements for the sampler or other engines
Something like the following might be better

- zynbanks (or better presets)
    - xiz
        - bank1
            - 0001-preset01.xiz
        - bank2

jofemodo, 2017-10-29:
zynbanks directory is ONLY for ZynAddSubFX.  The name is the default name used by ZynAddSubFX for tis banks. It could be changed, but it would break backwards compatibility with the UI, so i prefer to not change it by now. 
Anyway, each engine can have its own directory structure inside zynthian-data or zynthian-my-data.

mheidt, 2017-10-29:
and i don't want to develope handlers and pages for each engine, if a generic solution is possible.
thats why I am suggesting a generic presets folder.
You could create a symbolic link zynbanks from presets/xiz

jofemodo, 2017-10-30:
OK! You are right! Let's find the best way!
Let me think a little bit ... ;-)

mheidt, 2017-10-30:
Check out my branch...

jofemodo, 2017-10-30:
OK Mr. @mheidt !

I've created a soft link from "zynbanks" to "presets/xiz" and it works almost perfectly ... Thanks!! ;-)

I say "almost" because after installing a "xiz" file from Musical Artifacts it gives and error 500, although the file is correctly installed:

`ERROR:tornado.application:Uncaught exception POST /api/lib-presets (192.168.1.198)
HTTPServerRequest(protocol='http', host='zynthian.local', method='POST', uri='/api/lib-presets', version='HTTP/1.1', remote_ip='192.168.1.198', headers={'Accept-Encoding': 'gzip, deflate', 'Host': 'zynthian.local', 'Cookie': 'user="2|1:0|10:1507117823|4:user|8:cm9vdA==|a1d99cd25e423a16f282d8d1c5beed387206a4ee022e8822d8cf680027d9bff3"; advanced_view=1', 'Content-Type': 'multipart/form-data; boundary=---------------------------19248923612112156299778482076', 'Upgrade-Insecure-Requests': '1', 'Connection': 'keep-alive', 'Content-Length': '1420', 'User-Agent': 'Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:55.0) Gecko/20100101 Firefox/55.0', 'Accept': 'text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8', 'Referer': 'http://zynthian.local/api/lib-presets', 'Accept-Language': 'en-US,en;q=0.5'})
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1346, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/zynthian/zynthian-webconf/lib/presets_config_handler.py", line 98, in post
    }[action]()
  File "/zynthian/zynthian-webconf/lib/presets_config_handler.py", line 97, in <lambda>
    'DOWNLOAD': lambda: self.do_download()
  File "/zynthian/zynthian-webconf/lib/presets_config_handler.py", line 170, in do_download
    self.selected_full_path = self.revise_filename(self.selected_full_path, downloaded_file, file_type)
  File "/zynthian/zynthian-webconf/lib/presets_config_handler.py", line 190, in revise_filename
    existing_program_numbers.append(int(m3.group(1)))
ValueError: invalid literal for int() with base 10: ''
`
Regards,

jofemodo, 2017-10-30:
Regarding the search in Musical Artifacts, i see you search by "tag", but most of xiz files are poorly tagged and it's difficult to find something unless you write "synth" or some commonly used tag.
Would it be difficult to search by title/content?

Regards,

jofemodo, 2017-10-30:
More about Musical Artifacts integration ... 
What about the zynaddsubfx banks that are uploaded as "zip" files?

mheidt, 2017-10-30:
I take a look later. 
I did some refactoring. There is this musical-artifact class now. And it should take care of unzipping as well. Even if it is in a directory structure, I flatten it. But it won't be perfect right now.

Rgd. the search. I am not quite sure whether tags and description work like an OR. We should do some testing and maybe ask Lucas.
He is quite corporate and maybe he creates better tags as well.

I am working on sf2 banks. Like "Zynthian Piano" or "Zynthian Bells".
Should I share them in my Dropbox and you look at it before I upload them to musical-artifacts?

thanks :)

jofemodo, 2017-10-31:
Hi @mheidt !
I don't think you must ask me for publishing anything in Musical Artifacts. It's an open repository and no Zynthian's specific. In fact, everybody can share any file in MA and it will be available from Zynthian's library manager. And that's good! ;-)

Anyway, it will be a pleasure to test your SF2 fonts, and it's an honor that you call it "Zynthian" ;-) 

Thanks!

mheidt, 2017-10-31:
bug fixed. But musical artifacts doesn't offer other search options.

https://github.com/lfzawacki/musical-artifacts/wiki/API-Documentation

When I would be him, I would demand better tagging as well instead of fulltext search, which is just too slow.
And if you look at the result https://musical-artifacts.com/artifacts.json?formats=xiz, there is not much more information in description anyways.

mheidt, 2017-11-06:
They should be unzipped. If not, tell me the tag, so that I can test it.

jofemodo, 2017-11-06:
Hi @mheidt !

I've merged this branch and added the code to "zynthian-sys" to create the "presets" directory structure. 
Anyway, i still get the 500 error after installing a preset from "musical-artifacts", although the preset is correctly installed.
Also, it would be nice to capture the "enter" key in the search box. Normally when you type a string for searching, you push enter by instinct and expect that it works ...

Thanks a lot!!

jofemodo, 2017-11-06:
Forget the last comment, it was duplicated by error and i've deleted ;-)

mheidt, 2017-11-08:
can we close this?

mheidt, 2017-11-09:
ok, forgot about the enter key.
If you still find the 500, send me a screenshot. I need to know the files in the bank and the filename of the preset you try to upload.
When you finished pressing all olives to oil, could we work on the bug in zyncoder (pitchbend)?

jofemodo, 2017-11-09:
Of course! We will do. I think that i know where is the problem.
Can you check if the problem is present when no tuning adjust is done?

mheidt, 2017-11-09:
Lets discuss that in the other issue

mheidt, 2017-11-16:
can this one be closed?

jofemodo, 2017-11-17:
Installing from Musical Artifacts is working OK. No more 500 error.
I see you implement the ENTER key in the D&D branch, so lets close this issue ;-)

closed 2017-11-17 by jofemodo