zynthian/zynthian-webconf #24: WIFI Configuration: Network list is not shown

jofemodo, 2017-11-06:
Network list is not shown. Although the JSON array with the network list is not empty and is received from browser, the select box with the list is not populated/shown.

jofemodo, 2017-11-06:
I've fixed the problem: The returned data object is not an JS array, but an jquery's iterable object, so it must be iterated with $.each.

mheidt, 2018-01-06:
@jofemodo  
A little different problem, but related.
New original v2 kit, updated.
List doesnt show, because wlan is wlan1 and not wlan0.
Why is that?
Or do we need a webconf change, so that it takes the first wlan* ?

jofemodo, 2018-01-10:
In my v2 box everything is working OK. It takes the RBPI3's internal wifi device as wlan0 and show the network list. I'm using the last "Gorgona Next" SD image.

mheidt, 2018-01-10:
I have a fresh one as well. I had this noise issue, so that I took the latest.
Hence I check for other wlan as well!?
Iterating until sth is found?

jofemodo, 2018-01-10:
Are you using the RBPi3's internal WIFI device?
Anyway ... iterating could solve the problem and it's a simple solution ;-)

commit 2b06f08b (2017-11-06): Fix #24: Nettwork list is not shown.
The returned data object is not an JS array, but an jquery's iterable object, so it must be iterated with $.each.
closed 2018-01-17 by mheidt