zynthian/zynthian-webconf #25: Multiple File Upload

mheidt, 2017-11-08:
- unlimit timeout
- <input type="file" id="myFile" name="img" **multiple**>

mheidt, 2017-11-08:
I found this
https://stackoverflow.com/questions/25529804/tornado-mime-type-of-the-stream-request-body-output

I won't start until I got your opinion, @jofemodo

jofemodo, 2017-11-09:
I'm studing the problem ;-)
Give me some days ...

jofemodo, 2017-11-11:
It seems good stuff, although i have found this one too:

https://techoverflow.net/2015/06/09/upload-multiple-files-to-the-tornado-webserver/

that seems simpler. Anyway, if you implement the functionality, you should take the final decision. Your code, your rules ;-)

Regards ... and thanks a lot!!

jofemodo, 2017-11-11:
What about the client-side interface? What about drag & drop files? It will be soooo cool ;-)

jofemodo, 2017-11-11:
Forget my suggestion, as it loads the full request into the webserver memory. Your option seems the best suited ;-)

mheidt, 2017-11-13:
Using tornado.web.stream_request_body is a little trickier.
It needs a separate request handler, because:
HTTPServerRequest.body is undefined, and body arguments will not be included in RequestHandler.get_argument.

But when I just change the file input to multiple, enlarging the max_body_size in zynthian_webconf.py to 6GB, we would have a cheap solution.

Drag and Drop should already work? just drop the file on the file-input.

If we decide to use stream_request_body, we need a separate popup. One with a bigger dropzone :)

I could push the quick fix. I don't know how long the better solution will take.

mheidt, 2017-11-15:
you successfully applied the chinese action strategy of silence :)
only for soundfont upload, I implemented drag and drop with a bigger dropzone and @tornado.web.stream_request_body

Pls checkout gh25-dragNdrop and tell me, when I can apply it to presets.

mheidt, 2017-11-16:
presets upload done.

jofemodo, 2017-11-17:
Hi @mheidt!

I'm testing the D&D branch:

+ Enter key for searching is not working for me (firefox@linux)
+ File uploading is not working. The files uploaded are corrupted. It seems truncated ... I've tested with ZASFX presets and also with SF2 soundfonts. When copying the same files by scp it works perfectly.
+ Drag&Drop is not working for me. No event seems to be fired when dropping the files. No error is shown in the JS console.

The webconf tool log doesn't shown any error:

```
root@zynthian:/zynthian/zynthian-webconf# git checkout gh25-dragNdrop
Branch gh25-dragNdrop set up to track remote branch gh25-dragNdrop from origin.
Switched to a new branch 'gh25-dragNdrop'
root@zynthian:/zynthian/zynthian-webconf# systemctl stop zynthian-webconf
root@zynthian:/zynthian/zynthian-webconf# ./zynthian_webconf.sh
INFO:tornado.access:200 GET /api/lib-presets (192.168.1.198) 41.62ms
INFO:tornado.access:200 GET /api/lib-soundfont (192.168.1.198) 20.66ms
INFO:root:{'program': '', 'text': '00001', 'name': '', 'nodes': [{'program': '001', 'text': '001.zss', 'name': '', 'bankName': '', 'fullpath': '/zynthian/zynthian-my-data/snapshots/00001/001.zss', 'id': 1, 'bank': '00001'}, {'program': '002', 'text': '002.zss', 'name': '', 'bankName': '', 'fullpath': '/zynthian/zynthian-my-data/snapshots/00001/002.zss', 'id': 2, 'bank': '00001'}], 'bankName': '', 'fullpath': '/zynthian/zynthian-my-data/snapshots/00001', 'id': 0, 'bank': '00001'}
INFO:tornado.access:200 GET /api/lib-snapshot (192.168.1.198) 23.48ms
INFO:tornado.access:304 GET /api/lib-soundfont (192.168.1.198) 4.21ms
INFO:tornado.access:200 POST /api/lib-soundfont (192.168.1.198) 6.12ms
INFO:tornado.access:200 POST /api/lib-soundfont (192.168.1.198) 6.02ms
INFO:root:copy /tmp/tmpylbi5yir to /zynthian/zynthian-my-data/soundfonts/sf2/Electric DigiChrome Piano.sf2
INFO:root:copy /tmp/tmp5fg8k4zz to /zynthian/zynthian-my-data/soundfonts/sf2/TimGM6mb.sf2
INFO:tornado.access:302 POST /api/upload?redirectUrl=/api/lib-soundfont&destinationPath=/zynthian/zynthian-my-data/soundfonts/sf2 (192.168.1.198) 2226.72ms
INFO:root:closing /tmp/tmpylbi5yir
INFO:root:closing /tmp/tmp5fg8k4zz
INFO:tornado.access:200 GET /api/lib-soundfont?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/soundfonts/sf2/Electric%20DigiChrome%20Piano.sf2?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/soundfonts/sf2/TimGM6mb.sf2 (192.168.1.198) 9.57ms
INFO:root:copy /tmp/tmp9yalu63x to /zynthian/zynthian-my-data/soundfonts/sf2/bassynt.sf2
INFO:tornado.access:302 POST /api/upload?redirectUrl=/api/lib-soundfont&destinationPath=/zynthian/zynthian-my-data/soundfonts/sf2 (192.168.1.198) 7.39ms
INFO:root:closing /tmp/tmp9yalu63x
INFO:tornado.access:200 GET /api/lib-soundfont?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/soundfonts/sf2/bassynt.sf2 (192.168.1.198) 4.52ms
INFO:tornado.access:200 GET /api/lib-soundfont (192.168.1.198) 4.16ms
INFO:root:copy /tmp/tmp52_fkl1u to /zynthian/zynthian-my-data/soundfonts/sf2/mbwow_ATT.sf2
INFO:tornado.access:302 POST /api/upload?redirectUrl=/api/lib-soundfont&destinationPath=/zynthian/zynthian-my-data/soundfonts/sf2 (192.168.1.198) 9.59ms
INFO:root:closing /tmp/tmp52_fkl1u
INFO:tornado.access:200 GET /api/lib-soundfont?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/soundfonts/sf2/mbwow_ATT.sf2 (192.168.1.198) 4.43ms
INFO:tornado.access:304 GET /api/lib-presets (192.168.1.198) 4.65ms
INFO:root:copy /tmp/tmpyoadkrr8 to /zynthian/zynthian-my-data/presets/xiz/Musical Artifacts/0001-GoodBass.xiz
INFO:tornado.access:302 POST /api/upload?redirectUrl=/api/lib-presets&destinationPath=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts (192.168.1.198) 5.38ms
INFO:root:closing /tmp/tmpyoadkrr8
INFO:tornado.access:200 GET /api/lib-presets?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts/0001-GoodBass.xiz (192.168.1.198) 4.87ms
INFO:tornado.access:200 POST /api/lib-presets?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts/0001-GoodBass.xiz (192.168.1.198) 6.18ms
INFO:tornado.access:200 POST /api/lib-presets?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts/0001-GoodBass.xiz (192.168.1.198) 6.16ms
INFO:tornado.access:304 GET /bower_components/bootstrap/dist/css/bootstrap.min.css (192.168.1.198) 27.00ms
INFO:tornado.access:304 GET /bower_components/bootstrap/dist/css/bootstrap-theme.min.css (192.168.1.198) 4.44ms
INFO:tornado.access:304 GET /bower_components/bootstrap-treeview/dist/bootstrap-treeview.min.css (192.168.1.198) 3.25ms
INFO:tornado.access:304 GET /bower_components/seiyria-bootstrap-slider/dist/css/bootstrap-slider.min.css (192.168.1.198) 3.67ms
INFO:tornado.access:304 GET /bower_components/font-awesome/css/font-awesome.min.css (192.168.1.198) 4.40ms
INFO:tornado.access:304 GET /css/style.css (192.168.1.198) 4.04ms
INFO:tornado.access:200 GET /css/zynthian.css (192.168.1.198) 4.20ms
INFO:tornado.access:304 GET /bower_components/jquery/dist/jquery.js (192.168.1.198) 16.02ms
INFO:tornado.access:304 GET /bower_components/js-cookie/src/js.cookie.js (192.168.1.198) 2.45ms
INFO:tornado.access:304 GET /bower_components/modernizr/modernizr.js (192.168.1.198) 5.93ms
INFO:tornado.access:304 GET /bower_components/bootstrap/dist/js/bootstrap.min.js (192.168.1.198) 5.11ms
INFO:tornado.access:304 GET /css/default.css (192.168.1.198) 3.60ms
INFO:tornado.access:304 GET /bower_components/bootstrap-treeview/dist/bootstrap-treeview.min.js (192.168.1.198) 4.04ms
[... cut ...]
INFO:root:copy /tmp/tmpou_7sagf to /zynthian/zynthian-my-data/presets/xiz/Musical Artifacts/0001-GoodBass.xiz
INFO:tornado.access:302 POST /api/upload?redirectUrl=/api/lib-presets&destinationPath=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts (192.168.1.198) 5.22ms
INFO:root:closing /tmp/tmpou_7sagf
INFO:tornado.access:200 GET /api/lib-presets?ZYNTHIAN_UPLOAD_NEW_FILE=/zynthian/zynthian-my-data/presets/xiz/Musical%20Artifacts/0001-GoodBass.xiz (192.168.1.198) 10.86ms
INFO:tornado.access:304 GET /css/zynthian.css (192.168.1.198) 2.35ms
```

Regards

jofemodo, 2017-11-17:
Ahhh! And sorry for the silence ... jajaja! It's not "chinese strategy", only "brain overload" ;-)

mheidt, 2017-11-17:
chrome works, firefox doesn't :(

jofemodo, 2017-11-17:
Firefox doesn't work for the "normal" uploads (without D&D)?

mheidt, 2017-11-18:
Read this
https://css-tricks.com/drag-and-drop-file-uploading/

mheidt, 2017-11-18:
i pushed some changes. I couldn't reproduce the truncating though. I had a bug before i changed the upload. 
My sf2 files load.
Have you read the url? There is no common solution for it. IE11 behaves differently as well.
The new firefox might too. So I vote for chrome support only until the browser companies aggreed on one solution.
Firefox is far more complicated (ajax). With chrome I can simply pass the files array from the event to the input object.

mheidt, 2017-11-18:
Instead, I used the saturday to work on a progressbar.
And that was the most difficult part so far.
Before I submit the fileupload form, I establish a websocket connection.
I connect the websocket handler and the following requesthandler with a random number key in an application context dictionary.
Check it out and tell me what you think.

mheidt, 2017-11-19:
And now firefox should work as well.

jofemodo, 2017-11-20:
Did you try with something like this?

https://github.com/Shopify/draggable

You should install it with bower in the zynthian-webconf directory ...

Regards,

mheidt, 2017-11-20:
Don't know, where this could have helped. The Drag/Drop was not the issue.
Passing the dragged files to an input[type=file] is handled differently between browsers, so that you have to use ajax instead of normal submit.

jofemodo, 2017-11-20:
I've been testing your last changes:

+ ENTER key works OK with search function
+ D&D works OK for single and multiple files
+ Upload doesn't work. It seems a problem with the progress ba. From the JS console in firefox:

Firefox no puede establecer una conexión con el servidor en ws://localhost/api/upload-polling.  lib-presets:164:43

```
socket:onconnecting: connecting { target: <div>, isTrusted: false, eventPhase: 0, bubbles: false, cancelable: false, defaultPrevented: false, composed: false, timeStamp: 14919.939384, cancelBubble: false, originalTarget: <div>, explicitOriginalTarget: <div> }  lib-presets:278:1
socket.onclose: close { target: <div>, isTrusted: false, eventPhase: 0, bubbles: false, cancelable: false, defaultPrevented: false, composed: false, timeStamp: 14920.244041, cancelBubble: false, originalTarget: <div>, explicitOriginalTarget: <div> }  lib-presets:288:1
```

Regards,

mheidt, 2017-11-20:
Try soundfonts pls

jofemodo, 2017-11-20:
So yes, the D&D is not the problem now ;-)

mheidt, 2017-11-20:
I think, i missed something at presets. the hashkey that maps the websocket connection with the request.
As I said, the progressbar is tricky...

jofemodo, 2017-11-20:
Same result with soundfonts ...

mheidt, 2017-11-20:
could you post the server log?

mheidt, 2017-11-20:
There are 4 testcases bth.
Firefox: D&D or Select by Click
Chrome: D&D or Select By Click

Not all of them fail!?

jofemodo, 2017-11-20:
Anyway, remember that you can add JS libraries to the project (zynthian-webconf) using the "bower" utility, that makes easy to maintain and update the JS libraries in a project.

There are thousands of JS libraries available and most of times, using the right library is the best way of leading with browser compatibility issues, etc.

I tell you that because i've not seen you using it until now, and perhaps you don't know bower ...

Regards,

jofemodo, 2017-11-20:
The 4 cases fail in the same way, giving the same JS error. It seems a problem with the progressbar and the websocket ...

mheidt, 2017-11-20:
i installed websocket with bower

jofemodo, 2017-11-20:
OK! I've seen now ... sorry ;-)

mheidt, 2017-11-20:
I only tested locally on my linux machine. Guess, I have to test on the zynthian.
That chrome fails as well, is weird.

mheidt, 2017-11-20:
Maybe it has something to do with localhost and cross site scripting!

jofemodo, 2017-11-20:
With Chrome the JS error is this:

`WebSocket connection to 'ws://localhost/api/upload-polling' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED`

mheidt, 2017-11-20:
yes, seems its the cross site scripting...

jofemodo, 2017-11-20:
FYI, MOD-UI uses websocket for communicating with the web app and also with the zynthian UI, and i had not problem with it.

mheidt, 2017-11-20:
have a look then :)
I won't until tonight...

jofemodo, 2017-11-20:
The webserver log doesn't show nothing related with this. No "ws://localhost/api/upload-polling" request reaches the server ...

jofemodo, 2017-11-20:
Anyway, the websocket should be open at "ws://zynthian.local/api/upload-polling" ... not localhost ;-)

mheidt, 2017-11-20:
:) RIGHT. I knew, there was a TODO. LOL

jofemodo, 2017-11-20:
Exactly. I've tested changing the WS URL and it works OK both in firefox & chrome ;-)

mheidt, 2017-11-20:
I will check tonight and will make it dynamic, if nobody else did it :)

jofemodo, 2017-11-20:
This works OK:

```
function connectWebsocket(connectedDeferred){
	var url = window.location.href
	var parts = url.split("/");
	socket = new ws("ws://"+parts[2]+"/api/upload-polling");
```

Regards,

jofemodo, 2017-11-20:
OK! I've commited this change ;-)

Regarding the uploaded files, some problem remains yet ... 

+ Uploading 1 single file => file is corrupted every time
+ Uploading several files => the last n files are corrupted, depending on the size of files

It seems that the last chunk of data is not correctly received/saved.

mheidt, 2017-11-20:
Only happens in the real slow world, not on my linux machine.

I have to change the streamer part and will try this one
http://pythonhosted.org/tornadostreamform/

mheidt, 2017-11-20:
and how do we get pip3 packages installed on the zynthian?

mheidt, 2017-11-20:
I checked something in, but didn't test on a zynthian myself. I need to install the pip3 package manually...unless you tell me, that I should move my pip3 installed files into the project.

jofemodo, 2017-11-21:
Tell me what packages are needed (send me the pip3 commands ;-) and i will add it as a post-update script in the zynthian-sys repository.

mheidt, 2017-11-21:
pip3 install tornadostreamform

jofemodo, 2017-11-21:
It's done. Update your zynthian box and you will get installed the tornadostream package ;-)

mheidt, 2017-11-21:
in case you are trying..still failing
the pip installed class uses os.rename
File "/usr/local/lib/python3.4/dist-packages/tornadostreamform/multipart_streamer.py", line 161, in move
    os.rename(self.f_out.name, file_path)
OSError: [Errno 18] Invalid cross-device link: '/tmp/tmp632lmxzj' -> '/zynthian/zynthian-my-data/soundfonts/sf2/Test.sf2'

My first attempt would be using shutil.move

mheidt, 2017-11-21:
pls check it out

jofemodo, 2017-11-22:
Yes,"/tmp" is a tmpfs file system, so you can "rename" a file from this directory to a filename outside "/tmp", so you get the "cross-device" error message. You should "copy" the file. I suposse you already figured this out ;-)

Let me try your last changes ...

jofemodo, 2017-11-22:
OK! I've checked and now it's working like a charm! Congratulations!!
When i've some time i will take a look to your code. I want to understand .... jeje! ;-)

Do you think this is ready to merge with master and close the ticket?

Regards,

mheidt, 2017-11-22:
yep, i created a gh26 already.

It's not all my code though. Thx  László Zsolt Nagy. :)

jofemodo, 2017-11-22:
jajaja! Of course ...

jofemodo, 2017-11-22:
You mean this guy:

https://hu.wikipedia.org/wiki/Nagy_L%C3%A1szl%C3%B3_Zsolt

;-DDD

closed 2017-11-22 by jofemodo