zynthian/zynthian-webconf #26: Add Update Software and Udate Library in configuration menu'

videobelu, 2017-11-09:
If is possible to add in the config web page also the menu to Update Sofware an Update Library for an easy setup process!!!

mheidt, 2017-11-09:
it would be cool if the update process would give me a log and a result, so that the user knows, if the update was successful.
So maybe some unique success keyword, i can grep.
And rows like LOG: The message, that should appear in webconf

jofemodo, 2017-11-17:
OK! I will do it, but it's not a priority, so it will take some weeks ;-)

mheidt, 2017-11-17:
Where do you see it in the menu?
I can start with showing the whole sys.out

jofemodo, 2017-11-17:
I think "System" menu would be OK ... do you agree?

mheidt, 2017-11-17:
yes, one menu item, page with two buttoms?
Or one buttom and checkboxes for the different scripts?

mheidt, 2017-11-17:
the backup/restore...we need to think of how to backup the configuration...is it enough to add the env setting file to the list of files/folders, that we backup?

jofemodo, 2017-11-17:
+ Yes, an update "tab" with several buttons or checkboxes would be OK.
+ We should have:
  + a button for "system backup", with a pre-configured (not editable from GUI) list of directories/files
  + a button for "data backup", with a configurable list of directories

Regards,

mheidt, 2017-11-22:
I would like to take
the button out of 
default.css
input, select, textarea, button, div.slider.slider-horizontal {
	width: 100%;
}

we should "class" them with btn-block instead.

jofemodo, 2017-11-22:
Feel free to do it, but use less, please. I suposse you already use it, only to be sure  ... ;-)

mheidt, 2017-11-22:
Checkout gh26-update-software

mheidt, 2017-11-30:
@jofemodo 
Could you write a diagnosis script, that displays everything we need to know for all those support cases, when the ERROR-Screen on the zynthian shows up?

Should we call the menu item "Maintenance"?

jofemodo, 2017-11-30:
It's a good idea, @mheidt! I'm adding a specific task in zynthian-sys. We will see how integrate it with webconf when it's ready.

Thanks!

jofemodo, 2017-12-11:
Hi @mheidt!
Now that #gh27 is merged, do you have to refactorize this (#gh26)?
I can't remember where, but you tell me something about this ... ;-)

mheidt, 2017-12-11:
Jejeje, you are alive :)
I did the refactoring. But I didn't do a regression test.
And you have to link the diagnosis script.
systen_update_handler, row 41

mheidt, 2018-02-05:
Additional Reset to factory settings
- Restoring default MIDI setup from template in  “$ZYNTHIAN_SYS_DIR/scripts”.
- Delete default snapshot.
- Restore default GUI “look & feel” options.
- Restore “hostname” and system password to default?
- Delete WIFI configs?

mheidt, 2018-04-01:
An update must not destroy manual changes in /etc/network/interface

mheidt, 2018-05-05:
I added the code of the former branch directly into master, because rebasing was a pain.
Now there are two not-yet-implemented buttons:
Factory Settings and Diagnosis script,@jofemodo

jofemodo, 2018-08-12:
Can we close this?

mheidt, 2018-08-13:
Haven't checked...Did you implement the diagnosis script?

jofemodo, 2018-08-14:
No. But the diagnosis script has an open issue in zynthian-sys.

mheidt, 2019-02-28:
Why did you open it?
The code still has parts of the diagnosis script.
If you don't want that script (which I wouldn't like), I have to rebuild something

jofemodo, 2019-02-28:
Hi @mheidt!
Perhaps i'm not understanding.This "diagnosis script" is not for the zynthian itself but for the "zynthian update process", right?

If it's the case, then it's already implemented in the Zynthian-UI:
```
		error_counter=0
		for cmd in self.commands:
			logging.info("Executing Command: %s" % cmd)
			self.zyngui.add_info("EXECUTING:\n","EMPHASIS")
			self.zyngui.add_info("{}\n".format(cmd))
			try:
				self.proc=Popen(cmd, shell=True, stdout=PIPE, stderr=PIPE, universal_newlines=True)
				self.zyngui.add_info("RESULT:\n","EMPHASIS")
				for line in self.proc.stdout:
					if re.search("ERROR", line, re.IGNORECASE):
						error_counter+=1
						tag="ERROR"
					elif re.search("Already", line, re.IGNORECASE):
						tag="SUCCESS"
					else:
						tag=None
					logging.info(line.rstrip())
					self.zyngui.add_info(line,tag)
				self.zyngui.add_info("\n")
			except Exception as e:
				logging.error(e)
				self.zyngui.add_info("ERROR: %s\n" % e, "ERROR")

		if error_counter>0:
			logging.info("COMPLETED WITH {} ERRORS!".format(error_counter))
			self.zyngui.add_info("COMPLETED WITH {} ERRORS!".format(error_counter), "WARNING")
		else:
			logging.info("COMPLETED OK!")
			self.zyngui.add_info("COMPLETED OK!", "SUCCESS")
```
OK! I know it's not very robust nor elegant, but it's working quite well right now ...
You should adapt a little bit the code, but it's quite simple. Just detect the "ERROR" string ;-)
No need for detecting the "Already" string ... 

If you are talking about a zynthian-wide diagnosis script, that would analyze the Zynthian-UI log among others, it's already addressed by another task and it's not currently a priority because it's not easy to write a really useful and functional diagnosis script. Moreover, the "Log Viewer" is on the way and will allow to see easily what is going on when something fails, and will make easier for novice users to send error reports that could be useful for developers, without the need of accessing the console.

mheidt, 2019-02-28:
I was thinking of a generic diagnosis script especially for the cases when zynthian-ui is not starting up but showing a red error,
Something we can ask the use to press when he or she is asking for help.
the upcoming logging is different. 
1. it is not activated by default
2. zynthian-ui bound.

The diagnosis should expecially print information about alsa, jack, xdisplay or whatever could help us to identify an issue.

mheidt, 2019-02-28:
So its more a collection of useful shell commands than an intelligent analysis script!

jofemodo, 2019-02-28:
OK! I will do something really simple and we will improve it on the way ...

jofemodo, 2019-02-28:
I would prefer to put the "diagnostics" script in other place than the "Update" panel. Perhaps the dashboard, or in its own menu option ...

mheidt, 2019-02-28:
it's just so convenient. I programmed it generic, so that you just can add any script.
And in the end it's a script you only use, when you are new (and updated) or have updated and it failed.

mheidt, 2019-06-16:
I like to connect that one
https://github.com/zynthian/zynthian-webconf/issues/71
with this one.

We will have a script that runs creates the bug template and contains diagnostic information for us.
The user can edit this result by giving us more details and  create a github issue.

mheidt, 2019-06-25:
I close this as there is #71 now

closed 2019-06-25 by mheidt