zynthian/zynthian-webconf #27: Backup user defined items

mheidt, 2017-11-22:
We should have:
- a button for "system backup", with a pre-configured 
(not editable from GUI) list of directories/files
- a button for "data backup", with a configurable list of directories

jofemodo, 2017-11-22:
OK! Better in a new task ;-)

mheidt, 2017-11-22:
I am not quite sure how to do it.
The system backup is the existing solution.
Maybe we need to change the content of the file/directory-configfile. Removing the my-data?
But how should we do the data backup section?
A) a tree of my-data with checkboxes? 
B) the tree without checkboxes but drag&drop to the content of a second configfile?

I think, B ist best. You want to define general backup tasks like all presets and certain sf2 files. With A) you have the problem, that it has to be possible to select a child and not the parent. Hard to understand for a user.
A) is better for single download tasks, not sth you might want to do every week.
But then, this single download is better suited in the soundfont screen and not backup&restore.

Your turn to tell me doing C) :)

mheidt, 2017-11-25:
Help needed, @jofemodo  :)

I added the new drag and drop upload to the restore.
Additionally I want to display the result of the uploaded files in time like I did in gh26.
Don't merge gh26 yet, btw.
So when I tried to do both, the upload-progressbar websocket and the restore-log websocket, I realized, that you can't do multiple websockets over one port.

https://stackoverflow.com/questions/12498429/multiple-websockets

So now I have 3 options.
a) Start second application on different port for the upload
b) Create a Dispatcher Websocket which handles all server to client communication
c) Ignore it and be prepared that the on_message can receive data from different endpoints.

Btw. I had to change the upload as well. Before these changes, the upload component did a reload of the page. Now I have a callback javascipt function and the caller decides, if a reload is needed.

And please let me know, how you like the 3 tab approach.
I will work on the edit option in the second tab. A tree and dragging files or folders into the right div.

When this branch is merged, I will merge with 26.

jofemodo, 2017-11-27:
Hi @mehidt!

I like the b.) option for the socket, so we could have several uploads running in different browser tabs. OK, it's not a very common case, but if you ask me, i could imagine somebody uploading 8 GB of soundfonts in a browser tab and trying to backup the config in another tab. Crazy, but possible ;-)

I think it shouldn't be too difficult to implement, and would be a good thing to have. Think that once we have this dispatcher, we can have other processes "injecting" useful information about Zynthian status. Something similar to what MOD-UI does. 

Moreover, it could be the foundation for a Zynthian Web User Interface, replicating the hardware UI functionality. Wishful thinking ... ;-D

Anyway, yours is the final decission. Your code, your rules ;-)

Regards,

jofemodo, 2017-11-27:
Regarding your first question, i think that the default backup configuration should be OK for 90% of users. People that want to customize this, probably doesn't have problems editing the backup config file by hand, adding or removing directory paths.

System config, presets, snapshots and other "light" data should be included in backups by default and nobody should want to change this. We can have some common options like "include soundfonts", backup "include plugins", etc. that could be checked/unchecked and that's all. 

For bigger customization, editing the config file should be OK.

Regards,

mheidt, 2017-11-27:
Hi @jofemodo 

Actually I think, we should do A) and B).
The own application and a different port has the advantage, that you could really do normal business and uploads in parallel. Furthermore you can have different settings (max body size etc)

you would want the web-ui and webconf in one app? I thought of creating a web-ui project and use a different port.
Especially because the ui will be different than the webconf-ui, i guess...more big buttons...and touch slide areas.

Nevertheless having one Dispatch-WebSocket, that takes the result of the restore and update is a good idea. 

Regarding backup/restore. 
Now I have 3 tabs. System-, Data-Backup and Restore.
Is it possible to have a pure system back, that it is especcially backing up the webconf choices.

We might need to split the setenv script? One script that can be backed-up and restored without the risk of loosing new features you might add.
Setenv only contains the defaults, but the other new script holds the fields with priority?

The data backup would be soundfonts, presets and snapshots. They shouldn't be separated, because the snapshots depend on the soundfonts and presets.
When I think about it, we shouldn't have this tree. Complete or none. 

What do you think?

mheidt, 2017-11-28:
@jofemodo 
Do you happen to know how SSO is possible?
I don't want to leave the second upload_app unprotected.
Or this is the reason, not to do it :)

mheidt, 2017-11-28:
Seems to work...
Still have this
@tornado.web.authenticated

And in the worstcase we adjust this
self.set_header("Access-Control-Allow-Origin", "*")

mheidt, 2017-11-28:
@jofemodo 

I am going to need this

sudo pip3 install jsonpickle

jofemodo, 2017-11-29:
It's added to the update script in zynthian-sys. Just "update software" and the package will be installed ;-)

mheidt, 2017-11-29:
I have committed something. But only watch, don't touch :)
I still have too many log messages. But I wanted to share and would like to know your opinion.

Now it's B) only. Dispatcher websocket.
Especially the restore of a backup is interesting.
Progress bar for the upload and after that the log of the restore process itself.

I hope you can give some input and tell me what I could do better. 
I have to test on the zynthian still.
Regression of Soundfonts/Presets as well.

And when this is merged, I can refactor gh26.

Ah and my notes regarding the setenv script!

mheidt, 2017-11-30:
@jofemodo 
I am updating my box right now but I get this:
Downloading/unpacking jsonpicklepip3
  Could not find any downloads that satisfy the requirement jsonpicklepip3
Cleaning up...
No distributions at all found for jsonpicklepip3
Storing debug log for failure in /root/.pip/pip.log

only jsonpickle?

mheidt, 2017-11-30:
filenames, that are needed:
/zynthian/config/system_backup_items.txt
/zynthian/config/data_backup_items.txt

the former one can be deleted.

Now you have the GO to merge.
I installed jsonpickle manually and did a successful backup/restore test.

jofemodo, 2017-12-01:
OK @mheidt! I've fixed the installation command. I will double-check and merge.
You are making a great work with the webconf, Markus!! Thanks a lot! I love you !! ;-)

mheidt, 2017-12-01:
thx. Unfortunately it's so much fun, that i don't find the time to play with the zynthian itself :)

jofemodo, 2017-12-11:
Hi @mheidt! It's merged ;-)

mheidt, 2018-03-04:
the entries in
/zynthian/config/system_backup_items.txt
/zynthian/config/data_backup_items.txt

mustn't end with a *

mheidt, 2018-03-04:
And furthermore the backup is kind of useless right now. A complete data backup is too big.
Maybe we should exclude the soundfonts.
In this case, we need the whitelist of the remaining subdirectories in the data_backup_items.

mheidt, 2018-03-06:
gh27-backup-toc

If you add folders in the beginning of the backup_items text file and append a ^,
they are excluded.
In the zipfile there will be one or many .exclude files that contain all those excluded files.

This file will be considered by the restore.

jofemodo, 2018-03-07:
That's very useful, @mheidt!
What about a way of editing the "backup_item" files from the webconf? A raw editing tool should be enough ...

jofemodo, 2018-03-07:
The extra "/*" are removed from "backup_item" config files:

master 8c77596

jofemodo, 2018-03-07:
Merged gh27-backup-toc in master.

mheidt, 2018-03-07:
prevent restore when excluded files are missing and the raw editing is missing

mheidt, 2018-03-09:
Editing the file is possible. 
Two areas each. Whitelist (directories) and blacklist (files also with Browse button)
DIrectories cannot be selected with <input[type=file]>
The backup is creating a zip that contains a .exclude file which contains all excluded files.
These files are displayed during restore with a hint that the following files might be needed. (in snapshots f.e.)
In order to backup those, there is a backup button in the soundfont area as well.

jofemodo, 2018-03-13:
OK! It's merged ;-)

Thanks!

closed 2018-03-15 by mheidt

jofemodo, 2018-05-08:
Hi Markus!

Don't you think that backup screen is a little bit confusing?

[image]

1.) We could hide the 2 textareas and the filelist using the "advanced view" feature.
2.) Why the second textarea "Excluded directories or files" have a "Browse" button for choosing a local file? Is it really useful? Anyway you have to add the full path by hand ...

Thanks!

mheidt, 2018-05-08:
2) This is to exclude big files. If you don't have the browse button, you are blind unless you use ssh. which we don't want the normal user to do. That you have to prepend the directory manually is a technical limitation we can't surpass...or i can't :)

jofemodo, 2018-05-09:
But the "browse" button will open a local file dialog. You wont see remote (zynthian) files ...

jofemodo, 2018-05-17:
Hi @mheidt!

The "Save" button doesn't have too much sense without the textareas ;-)

Regards,

mheidt, 2018-05-17:
Now your are picky :)

jofemodo, 2018-05-23:
;)