zynthian/zynthian-webconf #30: Pianotec licence input

mheidt, 2018-02-07:
Should we create a new menu entry?
Engines?
  - Pianotec

One Textarea for the licence.
ZYNTHIAN_ENGINE_PIANOTEC_LICENCE

jofemodo, 2018-02-20:
Can i close this task?

mheidt, 2018-02-26:
Or you let it open until we have the final picture after we talked to modartt

closed 2018-03-05 by mheidt

jofemodo, 2018-03-12:
Hi @mheidt !

Zynthian-webconf is calling pianoteq binary in boot time and it remains in memory all the time:

```
root@zynthian:~# pstree
systemd─┬─agetty
        ├─avahi-daemon───avahi-daemon
        ├─dbus-daemon
        ├─dhcpcd
        ├─htpdate
        ├─jackd───6*[{jackd}]
        ├─sshd───sshd───bash───pstree
        ├─startx───xinit─┬─Xorg
        │                └─zynthian.sh───zynthian_gui.py─┬─gpio
        │                                                └─13*[{zynthian_gui.py}]
        ├─systemd───(sd-pam)
        ├─systemd-journal
        ├─systemd-logind
        ├─systemd-udevd
        ├─ttymidi───5*[{ttymidi}]
        └─zynthian_webcon───zynthian_webcon───Pianoteq STAGE

```
Regards,