zynthian/zynthian-webconf #31: Display content of selected Snapshot

mheidt, 2018-02-11:
For the time being, only the unformated json format.
Helps for debugging. Later we can display only parts of it in a fashioned manner.

mheidt, 2018-02-11:
@jofemodo 
Please check it out and tell me, what content of the json you want to have displayed more user friendly.

jofemodo, 2018-02-13:
I would like to see the list of layers (channel, engine, preset), similar to the list shown in the UI.

mheidt, 2018-02-13:
You expect me to understand the xml?
But you told guysoft
"The snapshot file format is textual, but writing one by hand wont be so easy "

:)

no problem...

jofemodo, 2018-02-13:
Hi @mheidt!
The snapshot file format is JSON. Not difficult to understand at all if you display it with the browser ;-)
Writing it by hand is another question ...

mheidt, 2018-02-13:
reading writing...same for me. javascript and python offers helper :p

and i am that generation, that uses "xml" as synomym for any text based data format :)

jofemodo, 2018-02-13:
Perfect! ;-)

jofemodo, 2018-02-20:
Perfect! I love it, @mheidt ! Thanks a lot! ;-)

mheidt, 2018-02-24:
I had a js error..could't parse preset_info. It contained a carriage return (backslash n)
Maybe you can check in the preset save, that a trim is used.

jofemodo, 2018-02-26:
Can you give me some extra details about the snapshot giving the error? Could you attach the file?

jofemodo, 2018-02-26:
I've checked the snapshot save function and no "new lines" nor "carriage return" is generated explicitly.
The snapshot JSON code is generated like that:

`json=JSONEncoder().encode(snapshot)`

where "snapshot" is a dictionary containing the snapshot structured data.

mheidt, 2018-02-26:
0k, don't waste time with it.
In the end, i updated the file manually.
When the snapshot-lib in webconf doesn't load, we know that we should check the snapshot json.

closed 2018-02-26 by mheidt