zynthian/zynthian-webconf #32: Offer items in recorded audio folder for download

mheidt, 2018-02-11:
https://discourse.zynthian.org/t/recorded-audio-folder-in-webconf/1611

Hi, i was wondering if it’s possible to add the recorded audio folder (maybe inside library section) for grabbing recordings directly from webconf instead of manipulating ftp client

@jofemodo 
where is that folder?

ivan1071, 2018-02-11:
hi, first of all thanks :) after that: the folder is in:
/zynthian/zynthian-my-data/capture
at least in mine...

mheidt, 2018-02-12:
ha, got you...you used the sd card. shame on you :)

mheidt, 2018-02-26:
We need to decide, how we  get the usb drives. 
Is there something already?
Should we install pyudev?
https://pyudev.readthedocs.io/en/latest/

import pyudev
pyudev.Context()
ctx = pyudev.Context()
list(ctx.list_devices(subsystem='usb'))

Reminder: Plugin MIDI USB adapter and stick.

mheidt, 2018-03-05:
@jofemodo 
Pls tell me, which framework I should use.

jofemodo, 2018-03-16:
Hi Markus!

The USB drive is automounted. You only have to check if it's mounted in /dev/media/usb0. Take a look to this: 

https://github.com/zynthian/zynthian-sys/blob/master/sbin/jack_capture.sh

Regards,

jofemodo, 2018-03-17:
Hi @mheidt !
The captures in the SD card (/zynthian/zynthian-my-data/capture) are not shown ...

mheidt, 2018-03-17:
Sorry @jofemodo,

only tested on local machine and made wrong assumptions.

jofemodo, 2018-03-19:
OK! It works now.
A little detail: it would be nice to see if a file is from the USB unit or the SD card. Naming is repeated by the recorder in both places ...

jofemodo, 2018-03-19:
Anyway, i've merged with master ;-)

mheidt, 2018-03-19:
Are different icons ok?

jofemodo, 2018-03-19:
Of course! Any element that allows to differentiate the origin would be OK ;-)

jofemodo, 2018-03-26:
Do you think we can close this task?
For me it's working OK ;-)

closed 2018-03-26 by mheidt