zynthian/zynthian-webconf #33: Audioinjector and webconf

mheidt, 2018-02-25:
When I take a fresh installation, the audioinjector zero is recognized, but no sound is routed out.
When I use the zynthian-configuration, turn up the volume and leave the other alsamixer settings the same, the sound is
audible.
When I change the config tool, so that only alsa master and capture are manageable, i am not able "to make it sound".
I asked Matt from audioinjector, if he knows an anser.
I want to know, what we have to do automatically when the audioinjector is selected.
And I want to display "Master" and "Capture" only.

mheidt, 2018-02-27:
Matt answered:

amixer set 'Output Mixer HiFi' toggle

Where do we put this?
Is there something like you did for graphics?

mheidt, 2018-02-27:
or i could hardcode in post request

if self.get_argument('SOUNDCARD_NAME') == 'AudioInjector':
			try:
				call("amixer sset 'Output Mixer HiFi' unmute", shell=True)
			except Exception as err:
				logging.error(format(err))

mheidt, 2018-03-05:
You can close this, when you are happy with it

jofemodo, 2018-03-07:
I'm happy with it ;-)

closed 2018-03-07 by jofemodo