zynthian/zynthian-webconf #34: Adding a wifi hotspot option

guysoft, 2018-03-10:
Hey, so following request of @mheidt [here](https://discourse.zynthian.org/t/buildroot-for-zynthian/1589/46?u=guysoft).

The wifi hotspot is based [on this guide](http://www.raspberryconnect.com/network/item/330-raspberry-pi-auto-wifi-hotspot-switch-internet).

I will try to add tomorrow the full instructions. Here is a start:

* to change the name you need to edit the name in this file:
/etc/hostapd/hostapd.conf
```
ssid=RPiHotN
```
So the command is:
```
export NEW_NAME=zynthian_new ; sed -i "s/ssid=.*/ssid=${NEW_NAME}/g" /etc/hostapd/hostapd.conf
```
* To change password:
```
export NEW_PASS=zynthian_new ; sed -i "s/ssid=.*/ssid=${NEW_PASS}/g" /etc/hostapd/hostapd.confx
```
And then restart the access point (still need to test that).

* turning on and off. I still need to test it:
```
systemctl disable autohotspot
```
And disable a cron job for ``/usr/bin/autohotspotN`` (Still need to test that)

guysoft, 2018-03-12:
Message from developer of the guide for the hotspot script regarding the DNS issue on ethernet:

> Hi Guy
> 
> From what i have seen the /etc/resolv.conf file will be updated on a network connection from eth0 or > wlan0 automatically or refreshed when the device is taken down and up.
> So in the context of the script if it doesn't generate when eth0 is connected, if you add in the the elif  > statement for eth0 then you can use
> 
> ip link set dev eth0 down
>  ip link set dev eth0 up
> 
> which should solve the issue.
> 
> In Stretch the dhcpcd service updates the resolv.conf, it should have the hooks in   /lib/dhcpcd/dhcpcd-hooks 
20-resolv.conf  this will be run when dhcpcd is started and updated.

> I don't have a spare SD available at the moment, when I have one available I will have a look at the image.

> Hope this is some help.

> Graeme

mheidt, 2018-03-18:
@jofemodo 
do you want me to store ACTIVE, NAME, and PASSWORD in envars, so that you do the business logic.
Or should webconf change the system files directly?

guysoft, 2018-03-18:
I would not store a wifi password in envvar. 
I think the best way would be to write three scripts that change the username and password, and then one for on or off. Then I can also ship them with the auto-hostpost module.

You can write them and commit them either to zynthian-sys or to CusotmPiOS.
I would vote to shipt it in CustomPiOS because if @foosel decides, for example, to add the auto-hostpost module to OctoPi, we would all be contributing to the same codebase. But I would go with what you prefer.

What ever you commit in this folder would be shipped on the filesystem if auto-hotspot module is selected in the build:
https://github.com/guysoft/CustomPiOS/tree/CustomPiOS/src/modules/auto-hotspot/filesystem/root

mheidt, 2018-03-18:
i need one read_hotspot_name as well.

mheidt, 2018-03-18:
i don't have your image in my zynthian anymore. could you send me a /etc/hostapd/hostapd.conf and confx?

guysoft, 2018-03-18:
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/auto-hotspot/filesystem/root/etc/hostapd/hostapd.conf

What is confx?

mheidt, 2018-03-18:
guess i have to double check your command, now that I have the file:

To change password:
export NEW_PASS=zynthian_new ; sed -i "s/ssid=.*/ssid=${NEW_PASS}/g" /etc/hostapd/hostapd.confx

I reckon, the ssid is wrong also

mheidt, 2018-03-18:
wpa_passphrase

guysoft, 2018-03-18:
Why are you adding x at the end?
wpa supplicant is from: https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/network/filesystem/boot/custompios-wpa-supplicant.txt

It gets renamed according to hostname of the OS.

mheidt, 2018-03-18:
you quoted the x above and used the same regexp. I assumed, its a special file...before i saw the file...

guysoft, 2018-03-18:
My bad,  no x.

mheidt, 2018-03-18:
I committed sth into the branch gh34-hotspot.

guysoft, 2018-03-18:
Neat, what needs to happen for this to be merged to master?

mheidt, 2018-03-18:
can you test it? if it's ok, @jofemodo  will merge it

guysoft, 2018-03-18:
Might have time totomorrow,  no promises.

guysoft, 2018-03-20:
Ok,
So tested.
I ran fetch

1. If you put a wifi access point with a space in the middle, it takes only the first word.
2. When you hit "save" the page refreshes to a dead screen, so perhaps it might be a good idea to display something using ajax before. Moreover, if you reconnect and then hit "try again" it tries to re-subbmit the last post, which was to change the wifi.
3. If you try to turn off the page you get redirected to a page that says: ``400: Bad Request``/
[image]. Wifi does not seem to turn off.

To turn the wifi off you need to disable the crontab job that runs every minute. I'd use something like [python-crontab](https://pypi.python.org/pypi/python-crontab).
I wrote something that uses it and you can see [an example here](https://github.com/guysoft/AlarmBot/blob/master/src/alarm_bot.py#L119) (but that example filters according to a comment, so you would need to do it without filtering.

guysoft, 2018-03-26:
1. Cron ships with linux. You have it already

2. I can do it, I was just with other distros but irs freeing up. 

3. Ssh only works on second boot bug has been fixed.

guysoft, 2018-04-09:
Still not working, i get 400: Bad Request when editing the wifi tab under "advanced" and trying to save the text file.

guysoft, 2018-07-14:
Hey, tested latest nightly, the error in https://github.com/zynthian/zynthian-webconf/issues/34#issuecomment-374584583 still is present.

mheidt, 2018-07-17:
I tried the CustomPiOS from 07/15  and I couldn't get it running.
There were several issues before I end up where I am now.
First of all, I expect this config setup to be the default and not the 2.8 with Prototype 4 anymore:
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_ENCODERS

@jofemodo  do you agree?

I could use the Wifi Hotspot and changed it via webconf, but I get a red ERROR page in zynthian-ui.

Ah, I don't get why I had to regenerate the keys in webconf before I could ssh into the machine.
Normally a  ssh-keygen -f "/home/markus/.ssh/known_hosts" -R 192.168.30.124 should be enough.

I think, the goal should be to have an image, that works out of the box with the default zynthian kit.

I could update the software via webconf, but this failed somehow over ssh:

 ./update_zynthian.sh 
./update_zynthian.sh: line 30: /scripts/zynthian_envars.sh: No such file or directory
Updating zynthian-sys ...
fatal: Not a git repository (or any of the parent directories): .git

Same kind of error, when I try to run zynthian.sh

pi@zynthian:/zynthian/zynthian-sys/sbin $ ./zynthian.sh 
./zynthian.sh: line 6: /scripts/zynthian_envars.sh: No such file or directory
xauth:  file /home/pi/.Xauthority does not exist

/usr/lib/xorg/Xorg.wrap: Only console users are allowed to run the X server
^Cxinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: unexpected signal 2
Couldn't get a file descriptor referring to the console

this explains the ERROR message on the zynthian screen.

mheidt, 2018-07-22:
blocked by
https://github.com/zynthian/zynthian-sys/issues/65

guysoft, 2018-10-14:
Hey,
Thought you might like to know that pisound also have a wifi hostspot enabled by default.
They also have scripts to enable/disable it:
Toggle script:
https://github.com/BlokasLabs/pisound/blob/master/scripts/pisound-btn/toggle_wifi_hotspot.sh

Enable script:
https://github.com/BlokasLabs/pisound/blob/master/scripts/pisound-btn/enable_wifi_hotspot.sh

Disable script:
https://github.com/BlokasLabs/pisound/blob/master/scripts/pisound-btn/disable_wifi_hotspot.sh

jofemodo, 2018-10-25:
I just fixed the CustomPiOS's  "autohostspotN" script that wasn't detecting correctly the configured wifin networks in ZynthianOS. I created a pull request ;-)

jofemodo, 2018-10-26:
I also fixed the webconf code that writes the wpa_supplicant configuration, so no CR will be added.
Anyway, i think is a good idea to remove the CRs when parsing the wpa_supplicant in "autohostspotN". Improves robustness.

guysoft, 2018-10-26:
Merge

guysoft, 2019-02-28:
Is the hotspot stable now?

Talks here are also relevant:
https://github.com/guysoft/OctoPi/issues/396#issuecomment-462670627

jofemodo, 2019-02-28:
Yes. It works perfectly OK. Perhaps it could be fine to have the possibility of disabling it ... but it could be done by simply calling systemctl, true?

mheidt, 2019-05-06:
We need to  have the hotspot checkbox. 
And a description, which tells the user the name and password of the hotspot.

mheidt, 2019-05-13:
I added the checkbox, but don't know how to enable/disable the feature

jofemodo, 2019-05-13:
OK! It's my turn ;-)

guysoft, 2019-05-13:
Don't be shy to PR scripts to CustomPiOS auto-hotspot module, if this stuff is in a module external to ZynthianOS it might also get code pouring in from other projects.

jofemodo, 2019-05-14:
OK! I'm thinking of modifying the "/usr/bin/autohotspotN" script ;-)

filipgoc, 2019-05-30:
I just want to say, this is really exciting stuff! Thanks for all your efforts. It's often very difficult to connect to networks in schools and various restricted networks; this will really enable a new demographic to use OctoPi. It's getting so close 👍

jofemodo, 2019-05-30:
OctoPi? This is Zynthian ... excuse me, i don't understand ;-)

jofemodo, 2019-05-30:
BTW, we switched to use our own script for wifi hotspot. We don't use the CusmPiOS "auto-hotspot" module anymore, although we took some fragments from it:

https://github.com/zynthian/zynthian-sys/blob/master/sbin/set_wifi.sh

guysoft, 2019-06-02:
OctoPi is another distro using CustomPiOS @filipgoc just opened an issue there documenting his attempts to get the auto-hotspot working on OctoPi. 
I just suggested he should test [set_wifi.sh](https://github.com/zynthian/zynthian-sys/blob/master/sbin/set_wifi.sh) which you wrote about in the last comment, to see if that could be meted in to CustomPiOS, and then back down to OctoPi.
This is a feature requested by other distros and it seems like you got it working.

filipgoc, 2019-06-02:
@jofemodo Sorry for the confusion; that was way too much of a shortcut I took there. I am still excited about the hotspot development. 

Thanks for the information about the new script.

jofemodo, 2019-06-03:
No problem at all, @filipgoc! Only trying to understand ... ;-)

I wrote the new "set_wifi.sh" script because normally we want to totally stop wifi activity on zynthian and enable only when needed, and in the way we want (Hostpost or WIFI client). For instance, you could want to enable the hotspot node although in the range of a configured WIFI network. This is not possible with auto-hotspot.

The "auto-hotspot" script is designed for a different use-case where you want your node becomes "on-line" ASAP, with minimal human interaction, prioritizing the connection to a known WIFI network.

mheidt, 2019-06-16:
Any work left? I tested it and it worked.

closed 2019-06-16 by mheidt