zynthian/zynthian-webconf #35: Secury in Webconf

jofemodo, 2018-03-13:
@SrMouraSilva posted:

https://discourse.zynthian.org/t/question-about-secury-in-webconf/1717

4 actions should be taken in order to secure the Zynthian:

+ Webconf: secure "post" API calls
+ Webconf: regenerate cookie authentication private key on first boot
+ System: regenerate SSH keys on first boot
+ Webconf: add button for "regenerate keys"

guysoft, 2018-03-13:
If we ship with ZynthianOS then I see no reason to maintain the ``regenerate SSH keys on first boot`` on the Zynthian side. Same reason you don't handle creation of a root user - Its already there.

SrMouraSilva, 2018-03-13:
Other point:

https://github.com/zynthian/zynthian-webconf/blob/master/lib/zynthian_websocket_handler.py#L92

From https://jsonpickle.github.io/

> Python library for serializing any arbitrary object graph into JSON.
> Warning: jsonpickle can execute arbitrary Python code. **Do not load jsonpickles from untrusted / unauthenticated sources.**

A strategy that I used is a manual serialization/deserialization.
* serialization: https://github.com/PedalPi/PluginsManager/blob/master/pluginsmanager/model/bank.py#L124-L134 ;
* deserialization: https://github.com/PedalPi/PluginsManager/blob/d0350aeb43607ef6aa6e5a666490a40485166a1d/pluginsmanager/util/persistence_decoder.py#L45

From what I saw quickly, apparently websocket is used only to indicate upload progress towards the Zynthian to the browser. If it is only this, you could ignore received data.

mheidt, 2018-03-13:
No, its used more than that. And it is planned, that a webui (browser based  zynthian-ui) is using that channel for all kinds of communication.
I need a generic de-/serialization framework. 
We are a trusted and authenticated source. I don't remember, but I might use it for musical-artifacts as well - I trust them as well.

jofemodo, 2018-03-13:
Not a big issue right now. The latest security improvements from:

dd1875c

should be enough.

Regards,

closed 2018-03-13 by jofemodo

SrMouraSilva, 2018-03-13:
> We are a trusted and authenticated source. I don't remember, but I might use it for musical-artifacts as well - I trust them as well.

@mheidt, [if I know the secure cookie](https://github.com/zynthian/zynthian-webconf/blob/a28c658a8e1dbd34b71b6fb671942e28859950e3/zynthian_webconf.py#L80), so I can fake an authenticated user and execute a script to drop a Zynthian.

guysoft, 2018-03-13:
I wrote a script that generates a secret on first run, and saves it in to an sqlite database using sqlalchemy. That's how I solved it in my apps. Its even more secure because it generates a binary stream and not only readable characters.

jofemodo, 2018-03-13:
Hi @guysoft !

The cookie secret is already generated by webconf and saved in a plain config file. If you want to regenerate, just delete the file and webconf will generate a new one. This solution is self-contained  and robust.

Currently, Zynthian use text files for storing configuration values. Using a SQL database for storing configuration parameters is a strategical decision. Before introducing a new technology it should be discussed ...

Thanks,