zynthian/zynthian-webconf #38: Pianoteq: uploading doesn't work when using drag & drop

jofemodo, 2018-06-20:
Uploading doesn't work when using drag & drop. The received 
The log says:

`INFO:root:Installing /tmp`

When using the file dialog, it works OK:

`INFO:root:Installing /tmp/KIViR.ptq`

jofemodo, 2018-06-20:
Ahh! The preset cache is not regenerated correctly, but don't worry, it's fixed in my current development branch and it should be on-line by tomorrow ;-)

mheidt, 2018-06-20:
So is there something to do for me?
Do you have the 6.2 files I can test it with?

jofemodo, 2018-06-20:
If you like, the uploading + drag&drop problem. It's your code, so probably you will fix it in 2 mins ;-)
Regarding the 6.2 binaries, i've not tested yet, so please, test it!!  I'm not sure of how the configuration is created depending of version. Perhaps the optimization options must be reviewed by @dcoredump  ...

Regards

mheidt, 2018-06-20:
When I do drag & drop the KIVir.ptq, it shows in the bottom line.
When I submit:
INFO:root:Installing /tmp/KIViR.ptq
WARNING:root:Directory /home/markus/.local/share/Modartt/Pianoteq/Addons already exists.
INFO:root:Moving /tmp/KIViR.ptq to /home/markus/.local/share/Modartt/Pianoteq/Addons
ERROR:tornado.application:Uncaught exception POST /api/sw-pianoteq (::1)
HTTPServerRequest(protocol='http', host='localhost', method='POST', uri='/api/sw-pianoteq', version='HTTP/1.1', remote_ip='::1', headers={'Host': 'localhost', 'User-Agent': 'Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Ubuntu Chromium/64.0.3282.167 Chrome/64.0.3282.167 Safari/537.36', 'Accept-Language': 'de-DE,de;q=0.9,en-DE;q=0.8,en;q=0.7,en-US;q=0.6', 'Connection': 'keep-alive', 'Content-Type': 'multipart/form-data; boundary=----WebKitFormBoundaryDrQFLN58cUNhRLqi', 'Origin': 'http://localhost', 'Referer': 'http://localhost/api/sw-pianoteq', 'Content-Length': '416', 'Cookie': '__lnkrntdmcvrd=-1; user="2|1:0|10:1529516049|4:user|8:cm9vdA==|107ba099caa4eaf5bc3a64323a779d6f1cdd0e45778dc8d513301e05e7996c67"', 'Accept-Encoding': 'gzip, deflate, br', 'Accept': '*/*'})
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 1509, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 2897, in wrapper
    return method(self, *args, **kwargs)
  File "/home/markus/Documents/zynthian/zynthian-webconf/lib/pianoteq_handler.py", line 67, in post
    }[action]()
  File "/home/markus/Documents/zynthian/zynthian-webconf/lib/pianoteq_handler.py", line 65, in <lambda>
    'INSTALL_PIANOTEQ': lambda: self.do_install_pianoteq(),
  File "/home/markus/Documents/zynthian/zynthian-webconf/lib/pianoteq_handler.py", line 115, in do_install_pianoteq
    self.pianoteq_autoconfig()
  File "/home/markus/Documents/zynthian/zynthian-webconf/lib/pianoteq_handler.py", line 206, in pianoteq_autoconfig
    info = get_pianoteq_binary_info()
NameError: name 'get_pianoteq_binary_info' is not defined

This is the second error, you were talking about?

So you have to look in the javascript log, what is going wrong in your case.

What browser are you using? I am using Chromium.

mheidt, 2018-06-20:
firefox didn't show an empty tmp path either

jofemodo, 2018-06-22:
Curious. It works now. I found some kind of ghost bug and i can't reproduce now.
Close the issue by now ... ;-)

jofemodo, 2018-06-22:
OK! I have reproduced the error. The error appears when dropping more than 1 file. I've tested with gh38-drag_and_drop_fix branch, but the error is still there.

mheidt, 2018-06-22:
Should we allow more than one file?
I think, this process is better one by one. 
Don't we expect a reboot after initial install before you install instrument patches?

mheidt, 2018-06-22:
When someone drops multiple files, only the first file is taken.
Could you solve the get_pianoteq_binary_info error and merge with master?

jofemodo, 2018-06-25:
Hi @mheidt !

I've tested your last changes and it works. It's merged with master now ;-)
Regarding the get_pianoteq_binary_info error, it works perfectly fine in my zynthian boxes. Probably it's a problem with your desktop development environment. Have you updated your local copy of zynthian-ui directory?

Regards,

mheidt, 2018-06-25:
Yes, sure.. that must be it...sorry :)

You can close this one then?

closed 2018-06-25 by jofemodo