zynthian/zynthian-webconf #39: Pianoteq: Add support for "Standard" and "Pro" binaries

jofemodo, 2018-06-25:
Come from this thread:

https://github.com/zynthian/zynthian-sys/issues/60

edwardjarchibald, 2018-06-25:
I made a quick hack to get Pianoteq Pro to install, via webconf, and in the process saw a need to make the install more generic.  Last weekend I worked out the approach which, when completed, will be version agnostic so, in theory, it will even work for an upcoming version 7 etc. as long as the command line interfaces don't change. I would also like to work out a means to always have all existing instruments/presets appear on the UI i.e. when a new instrument is introduced, I would like to have the menu automatically reflect that after an install.

I have a question for the community: Is there any reason to allow for the install and operation of more than one Pianoteq version at a time? If so, what is the reason? If there's a big Yes to this then I could work that out. Otherwise my tendency is to just have any newly uploaded version, whatever it is, fully replace the installed version. Of course this doesn't include replacing prefs, presets etc.

I got a little delayed because I have never written a line of Python code and I found it quite amusing the way that it cares about indents and what you use to indent with. I also found it amusing that I could not work on code in Eclipse, copy it from my working window, and paste it into the Python interpreter to try out stuff.  If you guys use a nice IDE to do Python coding/testing/debugging please let me know what it is. In any case I'll push my changes to a branch when I'm done - probably next weekend.

Jonnie-Cache, 2018-07-03:
I'm also interested in assisting with this. I'm just having trouble getting a dev environment set up (with emulation) on my Mac. All of the SSH/SFTP is driving me batty.

@edwardjarchibald, I would love to work with you on this. I have the exact same issue and it seems like the code does need to be generalized so that it will work with any version of Pianoteq.

I don't see any real reason to have more than one version of Pianoteq installed at a time. I understand that the Pro version may be more taxing on the CPU, but my hope is that it will work well enough on the 3B+.

In terms of IDEs, I have done some research and it seems that PyCharm is the IDE to beat. It looks pretty complex, but also very flexible. I've applied to get a free license from them since this is an open source project. But, in the meantime, they have a community edition available for free:

https://www.jetbrains.com/pycharm/download

jofemodo, 2018-07-24:
Hi @Jonnie-Cache!

Have you make some improvement in the installation procedure for "pro" version?
I would like to include it in master ASAP ... ;-)

Regards,

jofemodo, 2018-10-18:
Hi @Jonnie-Cache & @edwardjarchibald !

I've been working in a more generic Pianote installation process. It's almost ready:

+ update your zynthian software and reboot
+ try to install a new pianoteq binary package from the webconf

It should work with any Pianoteq product: Standard, Stage & Pro.

Currently there is a problem when detecting the licensed instruments for re-sorting the instrument list. It works with Stage, but doesn't work with Standard. I don't know if it work with Pro because i did't have it.

Could you test the Pro version and give me feedback?

Thanks!

jofemodo, 2019-02-17:
It's working OK with 6.4.1

closed 2019-02-17 by jofemodo