zynthian/zynthian-webconf #42: Log Midi Messages

mheidt, 2018-08-03:
I would write a WebSocket page that logs all incoming messages.

But how do I get those?

They should only be logged, when the webpage is open (aka a new  WebSocket Handler is registered).

What about filter. Does the webconf send the filter when it starts logging?
Or do we send everything and we filter with JS only?

tail -f based or direct callback function to the webconf?

jofemodo, 2018-08-08:
I have added a new "zynthianh_debug" service (update your software). For getting the logs you simply:

```
systemctl stop zynthian
systemctl restart zynthian_debug

```
After that, you can get the logs with:
`
journalctl -f -u zynthian_debug`

When finished, you should restart the normal service:

systemctl stop zynthian_debug
systemctl start zynthian

Tell me if this is enough ;-)

Regards

jofemodo, 2018-08-08:
Ahhh! Wait a moment, this is for logging Zynthian UI messages, but i see you would like to log "MIDI messages" ... and this is a completely different task.

First, you would like to see the messages in different "places" of the flow, mainly

+ Unfiltered
+ Filtered

The best way of doing this is creating a jack-midi "logger" client that we could connect to the different "output ports" of ZynMidiRouter (zyncoder's new jack client name). Currently there is not a "unfiltered" output port, so i have to create one. Until then, you can connect to the "main_out" port, that is the output used by most of engines.

Regards,

jofemodo, 2018-08-08:
Anyway, the Zynthian-UI log viewer feature could be interesting too, so perhaps you can open a new task for that and use what i've explained in my first reply.

Regards

mheidt, 2018-08-08:
I was thinking more of a feature for everybody. Especially when you want to use the MIDI filters or the Master Midi functionality.

Imagine you have this master keyboard and you don't know what is sent when you press a certain button.
Having a view would help a lot.

jofemodo, 2019-02-19:
@mheidt, Perhaps it's a good momento to implement this ... ;-)

mheidt, 2019-02-19:
Tell me, how I get the midi messages.
I could start by displaying
journalctl -f -u zynthian_debug

jofemodo, 2019-02-19:
Currently this work for python 2.7 on zynthian:

```
import mido
mido.set_backend('mido.backends.rtmidi/UNIX_JACK')
with mido.open_input("ZynMidiRouter:main_out") as inport:
    for msg in inport:
        print(msg)
```
We should install mido for python 3 ...

pip3 install mido

I will add it to the next update ...

jofemodo, 2019-02-19:
Also rtmidi is needed:

pip3 install rtmidi

jofemodo, 2019-02-19:
Using this approach we can give the option of selecting WHERE to plug the MIDI logger:

+ Device1
+ Device2
+ ...
+ ZynMidiRouter:main_out
+ ZynMidiRouter:net_out
+ ZynMidiRouter:ctrl_out
+ etc.

That is very convenient for debugging MIDI router related problems, developing new MIDI features, etc.

jofemodo, 2019-02-19:
Regarding the Zynthian-UI log viewer, we should open a separated task ... ;-)

jofemodo, 2019-02-20:
OK! I just recalled that we already create this:

https://github.com/zynthian/zynthian-webconf/issues/43

jofemodo, 2019-02-28:
Hi @mheidt !

I've added mido and python-rtmidi, so you can simply update and start using it for implementing this task, if you like ;-)

mheidt, 2019-02-28:
I start with this task, when the ui log is merged into master

mheidt, 2019-03-02:
@jofemodo 
Ok, you can play with the first draft in the new branch.
I refactored ui-log as well. the 5secs are gone.

AND I found a major bug now that I see the midi log :)
You remember my detuned-bug, which was gone suddenly.
Now that I have the panic button and can restart the ui, it's fixed quickly :)

BUT here we go:
You remember, I have my zynthian tuned to 442, which is a pitchwheel value of 321 or the like.
When you use the pitchwheel, it won't work anymore for this channel. After that it's sending a complete different pitchwheel event on every note-event. some value below 0, which proves my former observation that I had to tune it up to get back to 442.

When I change the channel, it's back to 321 until I play with the pitchwheel again.

jofemodo, 2019-03-04:
Hi @mheidt!

The MIDI-log work like a charm, but the UI-log seems to be broken in the gh42-midi-logo branch:

```
INFO:root:incoming ws message {'data': 'SHOW_DEFAULT', 'handler_name': 'UiLogMessageHandler'} 
DEBUG:root:action: SHOW_DEFAULT 
INFO:root:spawn_tail_thread
INFO:root:journalctl -f -u zynthian
ERROR:tornado.application:Uncaught exception in /api/ws
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/tornado/websocket.py", line 402, in _run_callback
    callback(*args, **kwargs)
  File "/zynthian/zynthian-webconf/lib/zynthian_websocket_handler.py", line 94, in on_message
    handler.on_websocket_message(decoded_message['data'])
  File "/zynthian/zynthian-webconf/lib/ui_log_handler.py", line 177, in on_websocket_message
    self.spawn_tail_thread(False)
  File "/zynthian/zynthian-webconf/lib/ui_log_handler.py", line 124, in spawn_tail_thread
    UiLogMessageHandler.logging_thread = UiTailThread(self.websocket, loop, self.get_process_command(debug_level))
  File "/zynthian/zynthian-webconf/lib/ui_log_handler.py", line 75, in __init__
    super(websocket, loop)
TypeError: must be type, not ZynthianWebSocketHandler
INFO:root:Client disconnected
INFO:root:stopping tail threads
INFO:root:stopping tail threads
``` 
Also, we have to avoid the "double scrollbar" when using "not-big-enough" displays. It's a nightmare to move up and down with 2 scrollbars (a window with textarea higher than the window).

I've found & solved this problem on several projects before, so let me try it ;-)

Regards,

mheidt, 2019-03-04:
Ok, I fixed it Montag,

jofemodo, 2019-03-04:
Can you conmutate between debug and normal modes? It fails all the time in my zynthian ...

jofemodo, 2019-03-04:
Please, pull before modifying. I solved the double scrollbar problem in both logs ...

jofemodo, 2019-03-05:
Pull again! I just pushed some changes ... ;-)

mheidt, 2019-03-05:
And what did you change in the ui-log, so that it worked again?
The diff is unreadable, BC you added tabs.

mheidt, 2019-03-05:
I follow pycharms code style, btw.
Maybe you can export a style design file, I can import.

jofemodo, 2019-03-05:
Hi @mheidt!

I didn't change ui-log, only fix the indentation. But i merged your last commit with the "retrial" counter.

Regarding the coding style, almost all the zynthian code uses tabs, so it would be nice to respect this. Mixing is a nightmare, and regarding Python, it doesn't allow to mix tabs and spaces from version 3.
For the rest, i try to follow PEP ... more or less ;-)

mheidt, 2019-03-05:
Pycharm uses PEP.
That's why I wonder, that it wasn't correct.
Don't you use pycharm?

jofemodo, 2019-03-05:
I suposse ;-) I don't use pycharm ... i prefer kate. Anyway, i don't plan to convert the zynthian code to space-indentation. Pep says that although is not the preferred indentation style, using tabs is acceptable when the base code use tabs. This is the case ...  It also says that mixing tabs and space in the same project is highly discouraged ;-)

jofemodo, 2019-03-07:
Hi @mheidt!
I'm really happy with the MIDI logger, it's really useful for locating MIDI-related problems and simplify a lot the debugging of filter rules and also zynrouter development.

However, there is an limitation that would be nice to improve. Currently you can only "sniff" one point in the MIDI chain (ZynMidiRouter:main_out), and it would be really useful to have the possibility of sniffing in any point (aka. jack MIDI-out port).

So ... what about a select with the list of available MIDI output ports? ;-) The code is almost done in the MIDI ports section ...

Regards,

mheidt, 2019-03-07:
Yes, I had this in mind as well.
But then I thought, that all midi IN is routed to Out anyways and decided to not do it.
But your case is different. 
Tell me, which options you need.
Or how I get all options.

jofemodo, 2019-03-07:
I would like to have a list with:

+ All enabled "MIDI-input devices" => This is the difficult part, as this sublist is dynamic
+ ZynMidiRouter:main_out => This is the default
+ ZynMidiRouter:net_out => Only useful to people using QMidiNet (aka @wyleu)
+ ZynMidiRouter:ctrl_out => Feedback sent to devices (i.e. for the Behringer Mötor faders)
+ ZynMidiRouter:ch0-15_out => same than main_out, but filtered by channel (0-15)

Some of this options are not very useful for normal users, but can be really useful for developpers or advanced users. Perhaps we could add use de "advanced" flag for showing/hiding  some of them:

+ ZynMidiRouter:net_out
+ ZynMidiRouter:ctrl_out
+ ZynMidiRouter:ch0-15_out

Regards,

jofemodo, 2019-03-07:
Ahhh! Perhaps a "filter by channel" could be implemented in the logger tool, avoiding listing so many options and adding extra control to the user.

jofemodo, 2019-03-07:
Of course, other filtering option in the logger tool could be useful, like filtering the "ACTIVE SENSE" messages that pollutes the log and makes difficult to follow the flow, etc. It's a never-ending history ... ;-D

wyleu, 2019-03-11:
Thanks for the name check!
I think we want to be a bit careful about overloading the functionality at this stage.

jofemodo, 2019-03-11:
OK! Added the select with the MIDI ports: 98a266e

wyleu, 2019-03-11:
Seems to work!! :-D

closed 2019-03-26 by jofemodo