zynthian/zynthian-webconf #43: Zynthian UI Log

mheidt, 2018-08-08:
I (@jofemodo) have added a new "zynthianh_debug" service (update your software). For getting zynthian UI to log the debug messages, you simply:

systemctl stop zynthian
systemctl restart zynthian_debug
After that, you can get the logs with:

journalctl -f -u zynthian_debug

When finished, you should restart the normal service:

systemctl stop zynthian_debug
systemctl start zynthian

mheidt, 2019-02-24:
I am going to implement it as UI Log (and #42 as Midi Log) under the Interface menu?
Or System? Which is getting quite big otherwise.

mheidt, 2019-02-24:
Do we really want a solution, that implies restarting the service?
Shouldn't we strive for a better solution?
OSC message that starts the logging or the like?

mheidt, 2019-02-24:
I need your help here I can't make it work with 2.7
Maybe if I could use asyncio.run with python 3.7, it could work.
Or you have an idea.

jofemodo, 2019-02-24:
Sorry Markus! I don't have plans to upgrade to Python 3.7 in 2019.
Anyway, you shouldn't be working with 2.7 ... currently zynthian use Python 3.4 ;-)

jofemodo, 2019-02-28:
Hi @Markus!

We only need to restart the service if we want DEBUG messages to be enabled. By default the zynthian service is logging ERROR & WARNING messages, but not DEBUG or INFO. Stopping the normal zynthian service and starting the zynthian_debug service allows to log ALL messages. That's all.

So, by using something like:

journalctl -f -u zynthian

you should be able to get the log messages stream for the zynthian service (ERROR & WARNING).

For having a detailed log, including DEBUG and INFO messages, the service should be restarted in debug mode (zynthian_debug service) and then:

journalctl -f -u zynthian_debug

will do the trick ;-)

I prefer to use this approach because generating log messages also consume CPU cycles and some memory. Zynthian resources are precious and shouldn't be used for logging purposes "by default", so the default logging is limited to ERRORS and WARNINGS ...

Kind Regards,

jofemodo, 2019-02-28:
Ahhh! Perhaps having access to the "webconf" log itself could be useful ... ;-)

mheidt, 2019-02-28:
could you check again.

I wouldn't waste time on the webconf log. 
1. you can't watch them live anyways...not the "-f"-tail way
2. they are only of interest when we have a 500 error, which are rare and easily reproducible.

jofemodo, 2019-03-01:
Hi @mheidt !
I've tested and the UI log works really well.
It failed to switch from DEBUG to NORMAL some time, but it's not really important. Perhaps the 5 seconds sleep is not the best choice and we should wait until the status of the service is "DOWN". I never did such a thing, so i don't know if it will work of it's possible to do such a thing at all ;-)

Regarding the textarea, it will be nice to auto-adjust its size using JS (jquery) for fitting the display size. If not, when using a small laptop or mobile device is difficult to move up and down. Anyway, perhaps this should be done in other sections too.

Finally, it would be nice to have a base class for all this "not-envars-listing" sections. It would avoid a lot of redundant code. I will add a task for doing this and refactorize the needed handlers.

Good Work!! Thanks a lot, my friend ... ;-)

jofemodo, 2019-03-02:
I just merged with master ;-)

jofemodo, 2019-03-07:
I think we could close this ...

closed 2019-03-07 by jofemodo