zynthian/zynthian-webconf #44: Jalv/LV2 configuration

mheidt, 2018-09-19:
What would the config page look like?

A LV2 submenu?

A list of all mod-ui plugins with a checkbox?

Where do I save the checked selection how?
Is there a jalv config file i have to manipulate?
Or do we create symbolic links pointing the engines from mod-ui to javl?

dcoredump, 2018-09-19:
Hi Marcus,

I think that looking for `lv2:SynthPlugin` inside the LV2 descriptions of the plugins may help to find all useful plugins for jalv. Maybe such a list can be generated and cached?

Regards, Holger

mheidt, 2018-09-19:
Do I understand correctly, that the plugin is stored twice? mod-ui and jalv need their own copies?
Is this necessary?

jofemodo, 2018-09-19:
@dcoredump, the python bindings include a "get_class()" for getting the class of plugins. You can do something like:
```

for i in world.get_all_plugins():
  if i.get_class().get_label()=="Generators":
    print("{} => {}".format(i.get_name(),i.get_uri()))

```

jofemodo, 2018-09-19:
@mheidt, there is no duplication of plugins. The same installed LV2-plugins are used for MOD-UI and Jalv.

mheidt, 2018-09-19:
the commit into branch jalv is only a ui draft. No action implemented yet.

mheidt, 2018-09-20:
@jofemodo 
Pls let me know, when "import lilv" should work.
Now I need to install lv2 plugins on my pc. world.get_all_plugins is empty.
That's why I want to test it on a zynthian.

jofemodo, 2018-09-21:
Hi @mheidt!

If you update your zynthian, the lilv python-binding extension should be installed in the right dir ;-)

Regards,

mheidt, 2018-09-21:
@jofemodo 
Now you have something you can test with.
Loading the page takes a lot of time, because I call world.load_all every time.

mheidt, 2018-09-23:
I am going to save the json of all plugins to a file all_jalv_plugins.json. When you update the plugins, you need to delete that file, so that it will be generated again.

jofemodo, 2018-09-26:
Hi @mheidt!
I've tested the new plugin configuration screen and it works like a charm. After fixing indentation and some little details, it's now merged with master.

closed 2018-09-26 by jofemodo