zynthian/zynthian-webconf #50: Layer alias

mheidt, 2019-01-15:
We could have an alias "names" that could be configured from webconf. Webconf should understand the snapshot format, but it's quite simple. Simply follow the rabit from here: 

https://github.com/zynthian/zynthian-ui/blob/4428d5b1f80c8d90b2e1d6a2ef06723cf72949a1/zyngui/zynthian_gui_layer.py#L581

So ... webconf would show a snapshot preview with the list of "root layer" (fxchains could be shown too), allowing to edit an alias for each one.

jofemodo, 2019-01-15:
Regarding the "snapshot previewer", it should use the same notation than Zynthian-UI, like that:

>$CHAN#$ENGINE_NICKNAME > $BANK/$PRESET

i.e.

>1#ZY > Bass/Wah Bass

Regards,

mheidt, 2019-04-30:
We can close this. I don't really need it.

closed 2019-05-02 by jofemodo