zynthian/zynthian-webconf #52: Improve jalv screen

mheidt, 2019-02-13:
- Refresh button on top
- Different tabs for different types
- Search

mheidt, 2019-02-17:
Pls check.
There is the little restriction that you can't make any changes, when you filtered the view.
If we want this too, the save algorithm has to be changed completely.
We would need a partial update.

jofemodo, 2019-02-18:
Hi Markus!
Have you tested this commit in a zynthian box?
It doesn't work for me and probably it doesn't for other people. The new "enum" class you have introduced can't be used as key for a dictionary...

mheidt, 2019-02-18:
yes!? it works on my zynthian box!
[image]

jofemodo, 2019-02-18:
Have you tried to "Search for new Plugins"?

mheidt, 2019-02-18:
try again.

mheidt, 2019-02-18:
but i can't explain, why the former worked on one of my machines.

jofemodo, 2019-02-18:
It's working OK now! 
Good work!! It improves quite a bit the LV2 selection screen ;-)

closed 2019-02-18 by jofemodo