zynthian/zynthian-webconf #55: can't connect to hotspots with UTF chars

Baggypants, 2019-03-28:
Zynthian can't cope with wifi hotspots that have UTF-8 chars, e.g. I have one that is set as '┓┏ 凵 =╱⊿┌┬┐' in zynthian webui it looks like '\xE2\x94\x93\xE2\x94\x8F \xE5\x87\xB5 =\xE2\x95\xB1\xE2\x8A\xBF\xE2\x94\x8C\xE2\x94\xAC\xE2\x94\x90'

If I try to set it up as a wifi the web-ui throws 500 errors and the zynthian-ui refuses to connect to any wifi.

jofemodo, 2019-04-25:
What about this, @mheidt? Probably is not very difficult to solve ...
Although is not very common, some people use "strange" chars for the SSID ...

mheidt, 2019-04-29:
yes, i will take a look in a couple of days. guess nobody wants to rename their hotspots :)

mheidt, 2019-04-30:
Sorry, but my router doesn't allow those characters, so that I cannot reproduce it.
@Baggypants  can you send us a log of the webconf, when it happens?

Baggypants, 2019-05-29:
Can you set up a wifi hotspot with a phone? both recent iPhones and Android phones allow you to use those characters. I'll try and get a webconfig log.

Baggypants, 2019-06-11:
Still an issue with Aruk

Baggypants, 2019-06-11:
It seems to be the output of iwlist
```
root@zynthian:/zynthian/zynthian-webconf/lib# echo $(iwlist wlan0 scan | grep ESSID)
ESSID:"TRF-T.net" ESSID:"TRF.net" ESSID:"TRF.net" ESSID:"TRF-T.net" ESSID:"\xE2\x94\x93\xE2\x94\x8F \xE5\x87\xB5 =\xE2\x95\xB1\xE2\x8A\xBF\xE2\x94\x8C\xE2\x94\xAC\xE2\x94\x90" ESSID:"BTHub6-89QW_EXT" ESSID:"BTHub6-89QW"
```

However with bash you can 'unescape' the output for the correct list

```
root@zynthian:/zynthian/zynthian-webconf/lib# echo -e $(iwlist wlan0 scan | grep ESSID)
ESSID:"TRF-T.net" ESSID:"TRF.net" ESSID:"TRF.net" ESSID:"TRF-T.net" ESSID:"┓┏ 凵 =╱⊿┌┬┐" ESSID:"BTHub6-89QW_EXT" ESSID:"VM3682880"
```
possibly python can unescape like this.

mheidt, 2019-06-11:
have been there. done that. But to no avail.

In the end, it is converted back for json.
I am one step further though.
Now, it's displayed correctly in the select list. (javascript)

But I couldn't make it work in python for the tab.
Changing the option value is not the solution, because you need the \x notation for the POST.

mheidt, 2019-06-11:
@jofemodo  
what is the deal with the new color blue of the Add network button?
When should buttons be blue now?

mheidt, 2019-06-12:
Please test it. I am not proud of the solution. It's rather something you come up with, when you gave up.

closed 2019-06-14 by mheidt

Baggypants, 2020-02-19:
Heyyyy, I tested it and it doesn’t quite work.

Adding the ssid using the webconf was fine.

You can’t stuff the escaped Unicode into wpa_supplicant. It has to be converted to hexadecimal.
https://raspberrypi.stackexchange.com/questions/41069/how-do-i-configure-wpa-supplicant-with-default-iphone-ssid

Once I changed the line to `ssid=e29493e2948f20e587b5203de295b1e28abfe2948ce294ace29490` then it let me connect