zynthian/zynthian-webconf #58: dwc_otg.speed=1

mheidt, 2019-04-15:
add 
dwc_otg.speed=1

to /boot/cmdline.txt

As Checkbox in Display?
And it should be checked automatically, when a USB display is selected?

jofemodo, 2019-04-15:
I think that Audio Config section is the right place, but don't check it automatically by now. It's not clear that it works with all devices, specially those supporting USB-2.

jofemodo, 2019-04-15:
The text should be something like: "Limit USB speed to 12Mb/s (USB-1)"
In fact, it's also interesting for MIDI-USB devices too, but i don't think we have to repeat the option in the MIDI tab ... or yes?

mheidt, 2019-04-16:
I added the checkbox ZYNTHIAN_LIMIT_USB_SPEED

First I thought, adding the dwc_otg.speed=1 to the SOUNDCARD_CONFIG.
But in that case I have to parse it everytime in order to know, whether it has to be checked and stripped from SOUNDCARD_CONFIG again.
Better would be using it in update_zynthian_sys.sh?

jofemodo, 2019-04-16:
OK! It's done ;-)

jofemodo, 2019-04-17:
But ... there is some problem with the checkbox in webconf. It can't be turned off. After saving the checkbox is again checked.

mheidt, 2019-04-17:
and now?

mheidt, 2019-05-02:
Are you sure, that you put it in cmdline.txt?

jofemodo, 2019-05-02:
Ups! My fail ... i put it in the "config.txt". I will change it ASAP!

closed 2019-05-02 by jofemodo