zynthian/zynthian-webconf #59: Webconf causes browser to consume cpu whilst idle

riban-bw, 2019-04-18:
When a web browser has zynthian-webconf open it idles as a cpu usage higher than other sites, e.g. on a netbook, surf uses 7% CPU on zynthian but only 3% cpu on bbc.co.uk (after initial load). Is there activity (web sockets?) that is raising the cpu load and can such activity be optimised?

mheidt, 2019-04-18:
Which page?
When I run it locally on my linux machine, I see a python process using 12% when I watch the UI Log.
But others don't show anything suspicious.
But this is locally. I have to see, what a real environment looks like.

riban-bw, 2019-04-18:
Sitting on the main home page.

mheidt, 2019-04-19:
Idle
[image]

Google
[image]

Webconf 1
[image]

Webconf2
[image]

I don't see anything strange

mheidt, 2019-04-19:
Idle
[image]

Google
[image]

Webconf 1
[image]

Webconf2
[image]

I don't see anything critical

riban-bw, 2019-04-21:
Here are my results using an Intel Atom netbook running Surf browser:

bbc.co.uk home page after fully loaded:
[image]

zynthian.local home page after initial boot with no engines or snapshots loaded:
[image]

Note the WebKitWebProces process that is working relatively hard for zynthian but not event showing (must be completely idle) for bbc.co.uk. I guess this is a Surf process to manage traffic and further assume it is working to handle websockets from zynthian.

jofemodo, 2019-04-25:
Hi @riban-bw!

I can't reproduce the "high load" problem ...  
Could you test with other computers & browsers?

Thanks

jofemodo, 2019-05-10:
Should we close this?

closed 2019-06-16 by mheidt