zynthian/zynthian-webconf #61: Add option for peakmeter/cpumeter

jofemodo, 2019-04-30:
Add a new option in the UI tabfor choosing between Peak Meter & CPU Meter. I would prefer a select with both options because it's more explicit. The envar name is:

ZYNTHIAN_UI_SHOW_CPU_STATUS

mheidt, 2019-04-30:
Did I understood correctly?
An Option labeled "Meter" with the values "Peak" and "CPU".
In case of CPU, ZYNTHIAN_UI_SHOW_CPU_STATUS will be set to true, false otherwise?

riban-bw, 2019-04-30:
I would prefer "Audio" rather than "Peak" to make it more obvious what it does - especially for non-native English speakers.

jofemodo, 2019-04-30:
Hi @mheidt !
You understanded perfectly. An options labeled "Meter" with two options: "Audio Level" and "CPU Usage" .
"Audio Level" sets the envar to False and "CPU Usage" to True.

mheidt, 2019-04-30:
Please test

jofemodo, 2019-04-30:
It works like a charm!

Thanks Markus!

closed 2019-05-02 by jofemodo