zynthian/zynthian-webconf #63: Feature request: Ability to configure external x-server as display

riban-bw, 2019-05-04:
For testing (and possibly for an actual instrument) users may display the zynthian gui on an x-server that is external to the target (RPi) platform. This may be occur implicitly with x-window forwarding or may be explicitly configured using DISPLAY=host:display. It may be advantageous to allow such an external x-window display to be configured via the web-conf. This may manifest as a Display Type of X-Window and Config of DISPLAY=host:display, e.g. DISPLAY=192.168.0.1:0.

I would find this useful for displaying the zynthian UI on an x-server running on an Android tablet (which works quite well) without having to ssh into the Zynthian to configure it.

To implement this manually:

- start XServer XSDL on Android device
- ssh to zynthian
- systemctl stop zynthian
- DISPLAY=android_ip:android_display /zynthian/zynthian-ui/zynthian.sh

mheidt, 2019-06-25:
Can this work?
http://www.chromium.org/developers/design-documents/aura

riban-bw, 2019-06-25:
@mheidt - sorry, I don't follow? The request is to add another display configuration so that we can use a remote X-server as our display. I anticipate adding an entry to the Display configuration in web-conf which will configure Zynthain to direct its output to the x-server. Now that I am more familiar with Zynthain and web-conf I could probably add a patch.

mheidt, 2019-06-25:
And my idea was to display it in a browser, so that you don't need the normal approach anymore.
Just a suggestion.

mheidt, 2020-02-08:
https://discourse.zynthian.org/t/cant-use-vnc/3407/41

riban-bw, 2021-07-05:
VNC works very well and is relatively simple for users to understand, especially with noVNC implemented presenting UI in web page. I will close this request.

closed 2021-07-05 by riban-bw