zynthian/zynthian-webconf #64: Ability to configure full colour scheme

riban-bw, 2019-05-05:
The web conf currently allows configuration of:
- Background Color
- Text Color
- Light Color
- Panel Background Color
There are several other colours used by zynthian. It would be advantageous to be able to configure all colours used by zynthian.

mheidt, 2019-06-16:
We need to collect the definition of "full colour scheme"

Beltshassar, 2019-09-05:
Would it make sense til use some kind of template engine? I guess that would provide a definition and maybe even a suite of ready made themes to go.
I've no real python experience, but a web search tells me that ttk is a template engine for tkinter. I don't know if we're using any of the native widgets though.

jofemodo, 2021-07-05:
This is been adressed by:

https://github.com/zynthian/zynthian-issue-tracking/issues/170

Closing!

closed 2021-07-05 by jofemodo