zynthian/zynthian-webconf #66: Cannot change password from web-conf

riban-bw, 2019-05-11:
- Navigate to http://zynthian.local/api/sys-security
- Enter new password in "Password:" field
- Enter same new password in "Repeat password:" field
- Click "Save"
Expected behaviour: root password updated.
Actual behaviour: root password remains unchanged.

jofemodo, 2019-05-13:
I can't reproduce this error. For me it's working like a charm ...

riban-bw, 2019-05-13:
I think this is due to some (special) characters not being escaped by the webconf, e.g. using an ampersand in the password seems to fail.

jofemodo, 2019-05-13:
OK! Please rename and redefine the issue ;-)

Thanks!

mheidt, 2019-05-13:
I think, it is especially &, because in that case the command is executed in the background.

jofemodo, 2019-05-13:
It should be solved now. Please test and confirm.

mheidt, 2019-05-13:
this is what i found:
If you use a shell other than bash, "echo" might not be a builtin command, and the shell will call /bin/echo. This is insecure because the password will show up in the process table and can be seen with tools like ps.

In this case, you should use another scripting language. Here is an example in Perl:

#!/usr/bin/perl -w
open my $pipe, '|chpasswd' or die "can't open pipe: $!";
print {$pipe} "$username:$password";
close $pipe

jofemodo, 2019-05-13:
Simpler is better. I can't see a real security problem in that. If someone can do a "ps", this person is logged in, so he is the root and can''t change the password to whatever he wants.. Also, bash is the default shell ...

IMHO it's OK like that.

Thanks!

mheidt, 2019-05-13:
but are you sure, that your fix solves a & in the middle? I would have expected quotes around user:password...and then the next bug, what happens, if a quote is in the password

jofemodo, 2019-05-13:
Ohhh!! Who use quotes in the password? Only somebody looking for bugs ... ;-D
In such a case, escaping the text would solve the problem.

riban-bw, 2019-05-14:
Remember this is the root user password for a system that is connected to a LAN. I would suggest that some users may want secure password and this may use unusual characters. I would expect that there is a mechanism already implemented (in almost every other password handler) to support all characters.

jofemodo, 2019-05-14:
Zynthian is only a poor synthesizer, not a server nor a laptop where you save personal data.
Anyway, currently all the characters should be accepted for system password ;-)

jofemodo, 2019-05-17:
Can i close this?

mheidt, 2019-05-18:
i wanted to test the double quotes still ... have to get my test  zynthian running first ... 
I think, webconf has to disallow the double quotes to make it perfect.

riban-bw, 2019-05-18:
Other characters also failing, e.g. dollar sign '$'.

riban-bw, 2019-05-19:
This function should work:

```
from crypt      import crypt
from subprocess import Popen, PIPE	
def set_password(password):
	p = Popen(['usermod',  '-p', crypt(password), 'root'], stdin=PIPE, universal_newlines=True)
```
I have tested with many characters including: !"£$%^&*()<>?,./:@~;'#{}[]_+-=|\`¬

riban-bw, 2019-05-20:
Doesn't work with backtick '`'.
When it fails it shows an error message with the password attempted which exposes someone's secret password to anyone who can see the screen until such time they can clear the screen.

jofemodo, 2019-05-21:
OK! Try now ;-)

closed 2019-05-21 by jofemodo

riban-bw, 2019-05-22:
That works now but it seems a little clumsy to escape each character we find is troublesome. There is a risk that other characters (which are not on my UK QWERTY keyboard but may on other users' keyboards) may exhibit the same issue.

I have submitted a patch in branch [simplify_password_handling](https://github.com/zynthian/zynthian-webconf/tree/simplify_password_handling) which simplifies the code and seems more robust. I think it also deals with the issue of clear text password being visible in process table because python encrypts password before calling shell command `usermod`.

jofemodo, 2019-05-22:
OK! The code didn't work because you forgot to import the "crypt" function. I've fixed, tested and merged.
Please, when submitting code, do basic testing before. Also when changes are very simple ... ;-)

Thanks!

riban-bw, 2019-05-22:
Gosh, sorry! Of course I did test this first, including the import but subsequently created the branch from my Windows laptop. I would normally copy the files to the RPi and test before committing but I failed to do this today. (I think I was traveling and was too eager to submit the patch.) I will ensure my normal test procedure is followed in the future.

commit 8f6586b7 (2019-05-13): Fix #66: Add quotes to channge password command for allowing special characters.
commit 260f2750 (2019-05-20): Fix #66 & #68: It should support all characters, including $, ", etc. If the command fails, an error will be shown.
commit 3a912d59 (2019-05-21): Fix #66: Escape backstick character when setting new pasword.
commit 09b8b2e6 (2019-05-22): Fix #66: Import crypt function & add logging error.