zynthian/zynthian-webconf #71: Direct issue reporting from webconf

riban-bw, 2019-05-20:
When I report an issue I need to find the right issue tracker (can be eased with browser bookmark) then provide information including details of the zynthian configuration. This is irksome and error prone and may lead to inconsistent reporting.

It would be advantageous to be able to click a link, e.g. in the webconf and be taken to the issue reporter with the zynthian configuration details pre-populated. If this proves challenging then it would be advantageous to be able to click a link, e.g. in the webconf and have this info presented, ideally added to the paste buffer. This would allow simpler and more consistent (and accurate) reporting.

The information should include:

- Software image, e.g. Aruk RC-1 2019-05-17-zynthianos-stretch-lite-1.0.0
- Software module versions
- Soundcard configuration
- Display configuration
- Wiring configuration
- Raspberry Pi version
etc.

It may be appropriate to simply add the zynthian configuration file to the issue report in which case a download link for this file would be useful.

mheidt, 2019-05-20:
I can take a look.

mheidt, 2019-05-20:
But provide a user/password and a test repo

jofemodo, 2019-11-29:
Moved to #108

closed 2019-11-29 by jofemodo