zynthian/zynthian-webconf #74: Web conf shows 2328 user preset

riban-bw, 2019-06-06:
Using a web browser, navigate to http://zynthian.local.
Log in if required.
Observe the quantity of user presets (2328 on my system).
Click on this number which is a link to http://zynthian.local/api/lib-presets.
Observer the quantity of presets (5 on my system).

**Setup:**
 - Aruk RC2
 - zyncoder: master (f672128) 
 - zynthian-ui: master (4a1e025) 
 - zynthian-sys: master (09e8d3e) 
 - zynthian-data: master (676c32a) 
 - zynthian-webconf: master (ae09848)

jofemodo, 2019-06-06:
I've fixed the preset-count code and now it works as expected. 
Anyway, webconf's preset management have to be fixed as it doesn't work as expected depending of the "engine". I think there is a issue that address that, mheidt?

closed 2019-06-06 by jofemodo

mheidt, 2019-06-06:
There is no task yet.  We defered it until the OSC interface provides all information that is needed to administer the presets.

commit 61986dca (2019-06-06): Fix #74: User-preset count in dashboard is not correct. It's not optimal ...