zynthian/zynthian-webconf #76: Add Record/Playback functionality to Capture Webconf

wyleu, 2019-06-10:
**Is your feature request related to a problem? Please describe.**
When using the Zynthian for playback/recording the having only the GUI interface to drive these operations can be limiting.

**Describe the solution you'd like**
Recording & Playback with and without looping of Audio & MIDI control functions are available from the Webconf Captures interface.

MIDI commands sent to driv CUIA. 

**Describe alternatives you've considered**
Often an issue when using a remote sequencer over MIDI Ethernet the ability to start and stop the sequencer and start and stop zyhnthian recording/playback from within the same environment ( MAC/PC/Linux/Android it doesn't matter; zynthian webconf is a responsive web interface) Leads to less 

Allows much workflow simplification when using zynthian for quick smash and grab recording of audio and MIDI.

**SOFTWARE**
zyncoder: master (f672128)
zynthian-ui: master (4a1e025)
zynthian-sys: master (09e8d3e)
zynthian-data: master (676c32a)
zynthian-webconf: master (61986dc)

jofemodo, 2019-06-10:
It's easily doable using CUIA, but having feedback about time position is not  currently supported ...

jofemodo, 2020-06-10:
Playback is currently supported by the webconf via browser HTML5 audio. Record is pending ...

jofemodo, 2021-07-05:
This is repeated on zynthian/zynthian-issue-tracking#265

closed 2021-07-05 by jofemodo