zynthian/zynthian-webconf #77: Copied gorgona snapshots don't appear in the webconf

Baggypants, 2019-06-12:
**Describe the bug**
I copied the contents of the directory `/zynthian/zynthian-my-data/snapshots` over from my old gorgona image to the new Aruk one, they appear in zynthian-ui but not in zynthian-webconf

**To Reproduce**
Steps to reproduce the behavior:
1. Copy across snapshot files
2. Go to webconf
3. Navigate to  *Library > snapshots*
4. observe no snapshots in the list.

Expected behavior**
You should see the list of snapshots

**Screenshots**

**Setup:**
 - SD-image: Aruk
 - Hardware Config: 
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA 

 - Software Version: 
zyncoder: master (f672128)
zynthian-ui: master (4a1e025)
zynthian-sys: master (09e8d3e)
zynthian-data: master (676c32a)
zynthian-webconf: master (61986dc) 

**Additional context**
Add any other context about the problem here.
I like pies.

mheidt, 2019-06-12:
could you zip the snapshots, so that I can try myself?

Baggypants, 2019-06-13:
Sure, However on the latest update they somehow even stop the zynthian-ui from loading so I renamed the snapshot folder.

```
SOFTWARE
zyncoder: master (80862b6)
zynthian-ui: master (c0b0517)
zynthian-sys: master (c2e285c)
zynthian-data: master (676c32a)
zynthian-webconf: master (e3f3a85) 
```
[snapshots.zip](https://github.com/zynthian/zynthian-webconf/files/3287601/snapshots.zip)

mheidt, 2019-06-13:
14-nosem3kr.zss contains PD Dream of '86 TUC.
the apostrophe breaks it.

mheidt, 2019-06-13:
test again please

closed 2019-06-14 by mheidt

Baggypants, 2019-06-14:
Confirmed fixed.