zynthian/zynthian-webconf #78: Git tag/branch management

mheidt, 2019-06-16:
Once we have more than one master branch and a stable tag f.e, we need the ability to change the branches/tags.

I was thinking of a new controller, which opens when we click on the branch (master) on the dashboard.
Now at each project  a select box with all branches/tags is provided.

mheidt, 2019-06-20:
I have a first draft ...in master...i created my local branch for it, but while testing I switched accidentally back to master...grrr :)

For now, it offers master and all tags. As there are none yet, you should only see master.
Create a test tag git tag test and you can play.

We should provide the tag "stable" and each release candidate could get one as well.

jofemodo, 2019-06-21:
I like the idea of having a tool for easily change the branch/tag for each zynthian repository.
I'm not sure of who is the target for that feature ... normal users, advanced users,  developers? All of them?

First, currently we don't have a "stable" or "development" tags ... 

Master branch is currently our "almost-stable" branch, and we use more-or-less per-issue branches for development: "Not trivial" fixes and new features are tested in a separated development branch and then merged with master. This branches should be deleted after closing, although currently it's not being done.

This is what we have and i think we should take profit of it, improving our efficiency as developers and easing the testing procedure for advanced users.

When we have a "stable" and "development" tag, it would be almost trivial to make the needed changes, but in this case, a simply global "checkbox" should be enough: stable / development

Regards

jofemodo, 2019-06-21:
BTW, i've deleted old development branches on every repository. I've conserved some historical branches, and the currently active branches ...

closed 2019-06-25 by mheidt