zynthian/zynthian-webconf #79: Ogg Download facility in webconf samples

wyleu, 2019-06-27:
**Is your feature request related to a problem? Please describe.**
Recorded samples are often loaded up into webpages as oggs . A wav file is preferred for 'normal' audio work but an ogg download facility will remove the conversion step increasing the likelyhood of use..
**Describe the solution you'd like**
A download ogg button.

**Describe alternatives you've considered**
At present I import to Audacity &export as ogg

wyleu, 2019-06-28:
What would you suggest as the wav->ogg tool?

*** Crazy side thought ***
Given it takes in a WAV, and I assume the processing is running on the zynth, could it be an audio routed to an output stage like the current system option ?

C Sound real time ogg encoding :-D

jofemodo, 2019-07-01:
oggenc seems to be the logic choice:

https://www.systutorials.com/docs/linux/man/1-oggenc/

Although ffmpeg should do the task perfectly too ;-)

jofemodo, 2019-07-01:
Ahh! You need to install "vorbis-tools" in your zynthian:

`$ apt install vorbis-tools`

mheidt, 2019-10-22:
@jofemodo is this installed in the current build already?

jofemodo, 2019-10-22:
Not currently, but it can be done for tomorrow ;-)
Also for updating stretch based systems ...

jofemodo, 2019-10-22:
But first confirm that it's really what we need ...

mheidt, 2019-11-30:
@jofemodo 
Please checkout branch gh79-ogg-download

wyleu, 2019-11-30:
[image]

Still downloads as wav . . .  

Not sure what the expected functionality is...

mheidt, 2019-11-30:
There is a new button....

wyleu, 2019-11-30:
Yes I saw that but it doesn't seem to do anything...

mheidt, 2019-11-30:
When you select a wav file, the button is enabled.
A Ogg file will be created and you find a new folder of Ogg files. 
Those files can be played in the browser.
Have you apt installed  the converter that was mentioned earlier?

jofemodo, 2019-11-30:
oggenc is not installed yet ... but i just commited the update package, so after updating zynthian, ti should work. I just tested in Aruk RC-3 and works like a charm ;-)

Regarding the option of streaming instead converting the file, it has the serious inconvenient of not allowing "transport control". Only play/stop ... so i prefer this option. The right solution should be solving the problems with wavs and then have buttons for "download as wav" and "download as ogg".

The branch is merged!

closed 2019-11-30 by jofemodo