zynthian/zynthian-webconf #80: liblo missing in buster image

mheidt, 2019-06-30:
I tested the buster image from 06/29.

root@zynthian:/zynthian/zynthian-webconf# ./zynthian_webconf.sh 
Traceback (most recent call last):
  File "./zynthian_webconf.py", line 35, in <module>
    from lib.dashboard_handler import DashboardHandler
  File "/home/pi/zynthian-webconf/lib/dashboard_handler.py", line 31, in <module>
    from lib.zynthian_config_handler import ZynthianConfigHandler
  File "/home/pi/zynthian-webconf/lib/zynthian_config_handler.py", line 28, in <module>
    import liblo
ModuleNotFoundError: No module named 'liblo'

mheidt, 2019-07-02:
it needs a 
sudo apt install liblo-dev

jofemodo, 2019-07-10:
It should be working on the next "green" nightly build ;-)

closed 2019-07-10 by jofemodo