zynthian/zynthian-webconf #81: Audio Injector Config needs Webconf config edit to 'activate' card.

wyleu, 2019-09-03:
**Describe the bug**
Audio Injector needs Webconf config edit to 'activate' card. When installing an Audio Injector card installation the card will not activate until a small edit ( addition of a space into the admin sections seems to work)

**To Reproduce**
Install Audio Injector Card
1. Install zynthian software.
2. Select audio hardware (Custom/Audio select Audio Injector Card)
3. Reboot machine
4. Login in to zynthian 
5. Run alsa Mixer 
6. Tab To Capture 
7. Select Capture L - R 
8. close alsamixer
9. Reboot.
10. No Audio output ( Audio Test) 
11. Select audio hardware (Custom/Audio select Audio Injector Card)
12. Edit custom settings 
13. Add space
14. Save
15. Reboot
16 Audio Output confirmed (Audio Test)

**Expected behavior**
An Audio INjector Card should play Audio Test upon installation

**Screenshots**
Not at the moment :-)

**Setup:**
- Not available.

**Additional context**
Been a little dance required for audio injector for a while now.
This processs does NOT confirm Audio input.

mheidt, 2019-09-03:
@jofemodo  moved the bit to update_synthian_sys.
Hence it's not an webconf issue anymore

mheidt, 2019-10-21:
I don't have any issues with my Audioinjector Zero and Ultra. Can we close this?

mheidt, 2019-11-24:
The latest build doesn't activate Audioinjector anymore. Ultra is ok.

jofemodo, 2019-11-25:
I've moved this code to "update_zynthian_sys.sh", that is executed every time config is changed.
I just improved the code for using the real device name instead of the card index. I hope this solve your issue.

Regards

mheidt, 2019-12-01:
You need to use the name of hw:xyz after -c, which is audioinjectorpi 
not the overlay name
audioinjector-wm8731-audio

jofemodo, 2019-12-02:
OK! It's changed! Please, confirm it works!

mheidt, 2020-05-26:
The latest 2020-05-25 doesn't activate audioinjector

mheidt, 2020-05-26:
when I call the script manually, Audioinjector works again.
It seems, that it is not called, when I update the settings and reboot? @jofemodo

mheidt, 2020-05-26:
Additionally to that I will see, what command is needed for the Ultra.

mheidt, 2020-05-26:
@jofemodo 
There are 2 issues. First it needs additional lines for Audioinjector Ultra. I don't know them yet. Testing takes so long, because I have to create new images quite often to get the initial state.
Secondly when I change the webconf by changing kit, audio, display, wiring in that order and only reboot in the end, the update_zynthian_sys.sh script is not called.
When I run it manually after the first reboot, TestAudio works.

mheidt, 2020-05-26:
could it be, that 
zynconf.save_config(sconfig, update_sys=True)
is called too early (before the reboot)
And that we rather need something like /temp/zynthian_reboot 
Something like /temp/zynthian_update_sys
?

mheidt, 2020-05-27:
I hope, I have my final test soon. (see Merge Request)
It seems, that the second reboot is not necessary. The jack error after reboot was not reproducible.
But calling the update-zynthian-sys.sh after the reboot is necessary.

mheidt, 2020-05-27:
@jofemodo 
The Merge Request contains the working commands.
Now I need your approval for the following:
When webconf is changing the soundcard, I will write a file /temp/zynthian_update_sys
When this file is found during reboot, update_zynthian_sys.sh will be called and the file will be deleted.
I need to know, where this could be done the best. (or you do it)

mheidt, 2020-05-27:
@jofemodo 
The merge request contains the working commands.
Now I need your approval for the following:
When webconf is changing the soundcard, I will write a file /temp/zynthian_update_sys
When this file is found during reboot, update_zynthian_sys.sh will be called and the file will be deleted.
I need to know, where this could be done the best. (or you do it)

jofemodo, 2020-05-28:
OK! I just added a new systemd service that will run the zynthian_update_sys.sh script when the flag file:

`/zynthian_update_sys`

does exist. The flag-file is deleted after.

**Note** that the flag-file is located in the root directory.

Please, @mheidt, modify the webconf for creating this flag-file when needed.

Regards!

jofemodo, 2020-05-28:
FYI:

https://github.com/zynthian/zynthian-sys/commit/e659cee6a0f7cf35f45ea9a380bcefb6043bf182

mheidt, 2020-05-29:
@jofemodo 
it doesn't work yet
I took the build of 27th, updated webconf, so that the file is written.
After the reboot, the file is still there.

Then:
systemctl start zynthian-config-on-boot
Job for zynthian-config-on-boot.service failed because the control process exited with error code.
See "systemctl status zynthian-config-on-boot.service" and "journalctl -xe" for details.
root@zynthian:/# systemctl status zynthian-config-on-boot.service
● zynthian-config-on-boot.service - Config-On-Boot Service
   Loaded: loaded (/etc/systemd/system/zynthian-config-on-boot.service; disabled
   Active: failed (Result: exit-code) since Fri 2020-05-29 19:59:05 BST; 20s ago
  Process: 1351 ExecStart=/zynthian/zynthian-sys/sbin/config-on-boot.sh (code=ex
 Main PID: 1351 (code=exited, status=203/EXEC)

May 29 19:59:05 zynthian systemd[1]: Starting Config-On-Boot Service...
May 29 19:59:05 zynthian systemd[1351]: zynthian-config-on-boot.service: Failed 
May 29 19:59:05 zynthian systemd[1351]: zynthian-config-on-boot.service: Failed 
May 29 19:59:05 zynthian systemd[1]: zynthian-config-on-boot.service: Main proce
May 29 19:59:05 zynthian systemd[1]: zynthian-config-on-boot.service: Failed wit
May 29 19:59:05 zynthian systemd[1]: Failed to start Config-On-Boot Service.

root@zynthian:/# journalctl -xe
-- The error number returned by this process is ERRNO.
May 29 19:59:05 zynthian systemd[1]: zynthian-config-on-boot.service: Main proce
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit zynthian-config-on-boot.service has e
-- 
-- The process' exit code is 'exited' and its exit status is 203.
May 29 19:59:05 zynthian systemd[1]: zynthian-config-on-boot.service: Failed wit
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit zynthian-config-on-boot.service has entered the 'failed' state with 
May 29 19:59:05 zynthian systemd[1]: Failed to start Config-On-Boot Service.
-- Subject: A start job for unit zynthian-config-on-boot.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit zynthian-config-on-boot.service has finished with a fail
-- 
-- The job identifier is 359 and the job result is failed.

jofemodo, 2020-05-30:
Hi @mheidt !

Please, update and test now. I fixed the problem .... i hope!

Thanks!

jofemodo, 2020-06-02:
Hi @mheidt !
Please, test and confirm that it works ;-)

mheidt, 2020-06-02:
I try, but it just takes too long. I am testing 06-01 btw, because your RC2 from 27th failed.

jofemodo, 2020-06-02:
Yes, this image doesn't contain these changes, but you can update and then test the changes.

mheidt, 2020-06-03:
@jofemodo 
No success.
Could it be, that after I changed kit, audio and display, the final change of the wiring is calling update_zynthian_sys already and deletes the file?
Maybe we shouldn't call update, when a we have a planned reboot?

mheidt, 2020-06-03:
yes, I made some tests. restart_ui removes the reboot flag.

jofemodo, 2020-06-04:
Then ... did your changes solved the problem? Is it working OK now?

mheidt, 2020-06-04:
I only checkeed 06-01 with an update call yet. But I want a complete image...testing 06-03 next

jofemodo, 2020-06-04:
Did you try "journalctl"? It should be there ...

mheidt, 2020-06-05:
I see
Jun 04 15:40:19 zynthian config-on-boot.sh[407]: /zynthian/zynthian-sys/sbin/config-on-boot.sh: line 7: /zynthian/zynthian-sys/scripts/zynthian_update_sys.sh

 and later
Jun 04 15:40:19 zynthian systemd[1]: zynthian-config-on-boot.service: Succeeded.

But I don't see
Configuring Alsa Mixer for AudioInjector ...

or even Updating System Configuration

Ha...
update_zynthian_sys.sh...not zynthian_update_sys....
Do you change it?

mheidt, 2020-06-05:
@jofemodo 
Have you changed the line?

mheidt, 2020-06-06:
@jofemodo 
nvm, I had the right and changed the file. I will test tomorrow again

mheidt, 2020-06-08:
@jofemodo 
No, still no success after first boot. It works, when I change the soundcard again and after the second boot.
But what I found after the first boot is, that webconf is set correctly, but in journalctl (yes, nearly end of file)
[image]

mheidt, 2020-06-08:
@jofemodo 
I guess the root cause will be this (a diff before and after the first boot after changing values in webconf)
root@zynthian:/zynthian/zynthian-webconf# diff /zynthian/zynthian-sys/scripts/zynthian_envars.sh /zynthian/config/zynthian_envars.sh 
24c24
< export ZYNTHIAN_KIT_VERSION="V3-PRO"
---
> export ZYNTHIAN_KIT_VERSION="Custom"
27,30c27,30
< export SOUNDCARD_NAME="HifiBerry DAC+ ADC PRO"
< export SOUNDCARD_CONFIG="dtoverlay=hifiberry-dacplusadcpro"
< export SOUNDCARD_MIXER="Digital,ADC,ADC Left Input,ADC Right Input"
< export JACKD_OPTIONS="-P 70 -t 2000 -s -d alsa -d hw:sndrpihifiberry -r 44100 -p 256 -n 2 -X raw"
---
> export SOUNDCARD_NAME="AudioInjector"
> export SOUNDCARD_CONFIG="dtoverlay=audioinjector-wm8731-audio\n"
> export SOUNDCARD_MIXER="Master,Capture\n"
> export JACKD_OPTIONS="-P 70 -t 2000 -s -d alsa -d hw:audioinjectorpi -S -r 44100 -p 256 -n 2 -X raw"
33,36c33,36

It seems, that first it will look in  /zynthian/zynthian-sys/scripts/zynthian_envars.sh ?
Although webconf is changing /zynthian/config/zynthian_envars.sh ?

mheidt, 2020-06-08:
@jofemodo 
Yes, I added the following line
Jun 08 10:30:38 zynthian config-on-boot.sh[368]: using /zynthian/zynthian-sys/scripts/zynthian_envars.sh

mheidt, 2020-06-08:
@jofemodo 
ZYNTHIAN_CONFIG_DIR is empty at this point.
Where is it initally set? 
Otherwise I would fix update_zynthian_sys so that I check file-existence with the hardcoded full path before looking at the backup position in scripts

mheidt, 2020-06-08:
https://github.com/zynthian/zynthian-sys/commit/f2dc0f0869e078e11634a7837a2992378e5f7d0f

mheidt, 2020-06-09:
@jofemodo 
You reverted the change!? For a reason or because you thought it belonged to the parallel change regarding zynthian_envvars?

jofemodo, 2020-06-10:
Hi @mheidt !

Please, confirm if it's working for you ...

mheidt, 2020-06-10:
No, it doesn't work.
You put back in a known bug:
Jun 10 08:44:51 zynthian config-on-boot.sh[371]: /zynthian/zynthian-sys/sbin/config-on-boot.sh: line 17: /zynthian/zynthian-sys/scripts/zynthian_update_sys.sh

update_zynthian_sys!

jofemodo, 2020-06-12:
OK! I see you fixed the typo ...
Then ... can we close this? ;-)

mheidt, 2020-06-12:
I can test on Monday Friday,

mheidt, 2020-06-14:
The good news:
The initial use case works now.
So if you get paid by closed tickets, you can close it :)

BUT.
The new usecase with /boot/zynthian_envars.sh doesn't

It seems, that there are two issues.
Initially I tried with the following file
export ZYNTHIAN_KIT_VERSION="Custom"

#Audio Config
export SOUNDCARD_NAME="AudioInjector"
export SOUNDCARD_CONFIG="dtoverlay=audioinjector-wm8731-audio"
export SOUNDCARD_MIXER="Master,Capture"
export JACKD_OPTIONS="-P 70 -t 2000 -s -d alsa -d hw:audioinjectorpi -S -r 44100 -p 256 -n 2 -X raw"

#Display Config
export DISPLAY_NAME="WaveShare 3.2B"
export DISPLAY_CONFIG="dtoverlay=waveshare32b:rotate=270,swapxy=1\n#dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=0,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900\n"
export DISPLAY_WIDTH="320"
export DISPLAY_HEIGHT="240"
export FRAMEBUFFER="/dev/fb1"

# Zynthian Wiring Config
export ZYNTHIAN_WIRING_LAYOUT="PROTOTYPE-4-WS32"
export ZYNTHIAN_WIRING_ENCODER_A="26,25,5,4"
export ZYNTHIAN_WIRING_ENCODER_B="21,27,7,31"
export ZYNTHIAN_WIRING_SWITCHES="107,23,106,6"

But then I get a GPIO 27 error
After diffing with a working file, I tried to add  the switch statements which were missing

export ZYNTHIAN_WIRING_CUSTOM_SWITCH_03__UI_LONG="NONE"
and the other 23
You could add those in the initial envars files, if you don't want to fix the error.
I think, it should be reproducible with a MCP23008 environment and delete those 23 lines after you configured it.

But besides the UI doesn't start yet.
I rebooted (I don't know yet if this is necessary) and changed the display and wiring in webconf forth and back.
Reboot again. I don't see any changes in the envars file after that step.
Maybe the custom steps of the waveshare32B weren't executed otherwise?
I am still investigating.

mheidt, 2020-07-24:
@jofemodo, I need your feedback. Should we move the wiring recompiling from webconf to script?