zynthian/zynthian-webconf #84: Sfz soundfonts download from musical-artifact never worked

lguyome45, 2019-09-04:
Trying to test as much as possible for zynthian Buster and Rbpi4, I have founded that sfz download does not work.

The system to download from musical-artifact, get the archive file, extract and remove all file except the file with the extension (sfz in your case).

The problem is sfz file is just a text file which call sample file in a directory, so removing all file make the soundfont not working.

I try to change the code to only do an extractall and remove nothing in case of sfz soundfont but there is some archive file with other directory.
For example, wintage kit https://musical-artifacts.com/artifacts/617 the directory structure is like:
|---Vintage Kit
|   |---Vintage Kit.sfz
|   |---.DS_Store
|   |---samples
|      |---All wav files
|---__MACOSX
|   |---Vintage Kit
|      |---._Vintage Kits.sfz
|      |---.__.DS_Store

We only need Vintage Kit.sfz and the directory samples, the problem is this archive have this structure of directory and other file can be different.

jofemodo, 2019-09-05:
I've a partial fix. I say partial because SFZ is a complex format and you will see that not all SFZs you download will work with Zynthian/LinuxSampler.

jofemodo, 2019-09-18:
Hi @lguyome45 !
Could you test this?

lguyome45, 2019-09-18:
Hi Jofemodo,
Sorry I miss to send report.
I make few test and like you say, there is some soundfont working and other not.
But for now, it is better, if you need some sample of working and not working soundfont, I can later send url of different situation.

jofemodo, 2019-09-19:
OK! I close this thread because now it works "as implemented", although implementation should be improved ;-)

closed 2019-09-19 by jofemodo