zynthian/zynthian-webconf #92: Buster / Audioinjector Ultra / Raspi4

mheidt, 2019-10-09:
I had a running system. An early buster image.
Once I updated, it didn't start anymore.
I took the image 10-08 and started new.
I could configure it via webconf 
But it still fails

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
libjackpeak failed to start jack client: 17
convert-im6.q16: unable to open image `/zynthian/config/img/zynthian_logo_error_ip.png': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: WriteBlob Failed `/zynthian/config/img/zynthian_logo_error_ip.png' @ error/png.c/MagickPNGErrorHandler/1641.
/zynthian/config/img/zynthian_logo_error_ip.png: image not found
didn't find evidence of prior run.
  Building XImage...done
^Cxinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2

What can I do to get more information?

jofemodo, 2019-10-09:
Hi @mheidt !
I found a problem with reboot & poweroff code in webconf that prevented it to work when zynthian-UI can't be started. I've solved and now it should work OK.

closed 2019-10-09 by jofemodo

mheidt, 2019-10-09:
do you know, why the former solution doesn't work anymore?

commit eb16b0c8 (2019-10-09): Fix #92: Fix poweroff and reboot controllers.