zynthian/zynthian-webconf #94: Save snapshot as default

mheidt, 2019-10-20:
https://discourse.zynthian.org/t/minimalist-zynthian-to-begin/3345/10

How should we design it?
2 new buttons Save as Default and Save as Last State

Or should we enhance the Bank with ("System")
and and Program with ("Default", "Last_State") and hack around the bank/prognum?

jofemodo, 2019-10-20:
Hi @mheidt !
First of all, i've refactorized and enhaced a little bit the snapshot manager in the last days. Don't forget to pull before changing ;-)
Regarding the implementation of this task, i prefer the 2 buttons solution. More simple and easy to implement.

Lemonmeth, 2019-10-20:
> 
> 
> Hi @mheidt !
> First of all, i've refactorized and enhaced a little bit the snapshot manager in the last days. Don't forget to pull before changing ;-)
> Regarding the implementation of this task, i prefer the 2 buttons solution. More simple and easy to implement.

Yes me too

mheidt, 2019-10-21:
open for testing

Lemonmeth, 2019-10-21:
How do i test?

mheidt, 2019-10-22:
Update your Zynthian. And use the new buttons in the snapshot library Tuesday,

Lemonmeth, 2019-10-22:
Works great for me!

closed 2019-10-22 by mheidt