zynthian/zynthian-webconf #96: Zero based Bank / Snapshot

mheidt, 2019-10-25:
In order to make the bank and program change work, it needs to be zero based

mheidt, 2019-10-25:
Could you check, if somewhere else 00001 is defined as first bank?
It has to be 00000 now

jofemodo, 2019-10-28:
Hi Markus!

Please, take a look to:

https://github.com/zynthian/zynthian-ui/blob/90574f56804897d318f54159a453464043439c5c/zyngui/zynthian_gui_config.py#L313

and this:

https://github.com/zynthian/zynthian-ui/blob/90574f56804897d318f54159a453464043439c5c/zynthian_gui.py#L1038

As you probably remember, we implemented the possibility of using relative (up/down) program/bank change. We defined the "Roland de-facto standard" and also allowed customizing the up/down commands.  The "program/bank" base (0/1) is adjusted internally depending on this feature.

So, currently the base program/bank numbers assigned from webconf's snapshot manager are "re-mapped" using this internal "base".

Please, think a little bit a bout all this and let's try to find a simple and coherent solution. I don't know if these features are being used by someone, perhaps we should ask the community ...

Thanks a lot! ;-)

mheidt, 2019-10-28:
I know all this, @jofemodo .
After Micki had problems (we have a german pm convo going on), I was testing it last week and made some changes in the webconf.
There are still some major bugs.
One is, that the default bank in the system has to be 0000 and not 0001.
Please check, where in ui/system it has to be adjusted.

commit d9b13198 (2019-10-28): Fix #96 & zynthian/zynthian-ui#167: Fix master program/bank change for snapshot loading.
commit 30d3cc4e (2019-12-21): Fix #96: Upload soundfont shows error in webconf.
Add try/catch for removing the uploaded file.
closed 2019-10-28 by jofemodo