zynthian/zynthian-webconf #101: flexible key bindings for QWERTY or numerical keyboard or keypad

hal80, 2019-11-06:
According to https://discourse.zynthian.org/t/control-your-zynthian-with-a-qwerty-or-other-keyboard/3057, functions of zynthian graphical user interface can be accessed by pressing according keys on a text/QWERTY keyboard. These key bindings are currently hard coded.
Request: It would be nice if these bindings where to be defined in a text file that can be edited by any user. This would also give the possibility to define key bindings for other keyboard layouts, such as numerical keypad.

jofemodo, 2019-11-06:
Hi @riban-bw !

This task mainly touch webconf, although some work is also needed in ui.
The place for the config file is "/zynthian/config" directory, that is addressed by the envar "ZYNTHIAN_CONFIG_DIR".
It could be interesting to have several mappings, including some "standarized" ones (i.e. "numerical-keyboard"), so perhaps it should be better to use a subdirectory "key-bindings" or so.
Regarding the format, most of configuration is done with environment vars, but currently JSON is used for snapshots and YAML is used for PD config files. Feel free to use any of them. We should avoid to add new formats unless it's strictly needed. Personally, i would use YAML for this ;-)

riban-bw, 2019-11-07:
Okay - I have implemented this using (my first foray into) yaml. Config file is under $ZYNTHIAN_CONFIG_DIR/keybinding with filename = name of binding configuration with extension .yaml. I used similar technique as used by pure data engine. I have committed the change to zynthian_gui_selector.py under branch https://github.com/zynthian/zynthian-ui/tree/feature_configurable_keybinding. I am away from home so cannot test this. If one of you feels brave you could test this and let me know if it continues to work as expected, i.e. default keyboard mapping works. I expect there may be bugs as I have been unable to test the changes, am not experienced with yaml (although it is abstracted from me) and am not a native Python programmer. I also am coding in a text editor with limited syntax checking... fingers crossed though! When I get this bit working the next step will be to implement the configuration editor in webconf which I may need a grown-up's help. We can test the implementation without webconf changes by manually creating and editing default.yaml which is loaded at init. If requested config file is missing then hardcoded default is used.

jofemodo, 2019-11-07:
Hi @riban-bw !

Nice implementation and nice to use yaml. It's very light, powerful and easy to edit by hand.
As i see you are highly motivated, perhaps it's a good time to improve your Python skills a little bit, jeje!! ;-)

The zynthian_gui_selector is a base class that is used by a good amount of child classes. When zynthian-ui is running, you have about 15 instances of zynthian_gui_selector. With the current keybinding implementation, the config file will be loaded 15 times, one by instance. Also, when loading a keybinding, you would need to call every instance's load method or restart the UI, etc.

The keybinding functionality should be implemented at "class-level". The load & save functions should be "class" methods and the variable containing the the configuration, a class variable.

Also, i would recommend to use the "zynapi_" prefix for naming the methods, as these methods should be callable remotely (i.e. from webconf).

As you probably know, i just implemented the first "zynapi" methods for allowing the new preset manager to interact with zyngines. You should take a look ...

https://github.com/zynthian/zynthian-ui/blob/0703e2c7f6445fbf82e7be149c81567d25e12bac/zyngine/zynthian_engine_zynaddsubfx.py#L459

The implementation is similar in the other zyngines. Ahh! Don't miss this little detail in the zyngine base class:

https://github.com/zynthian/zynthian-ui/blob/0703e2c7f6445fbf82e7be149c81567d25e12bac/zyngine/zynthian_engine.py#L557

This "get_zynapi_methods" method allows to get the available zynapi methods, so you can adapt the GUI options, for instance.

This is just the beginning, but i'm very satisfied with the result, so i will be following this path in other areas too: snapshots, layers, etc.

Currently "zynapi" methods are accessed from webconf by loading the python libraries, but i plan to integrate a webserver (tornado) in the UI, making the methods available using a HTTP REST interface. An OSC interface could be implemented easily too. Etc.

Regards,

riban-bw, 2019-11-07:
Ha! It is not so much my Python skills,  it is more a lack of familiarity with the Zynthian architecture. I must admit that I find the code to be a bit of a spider's web - it takes me time to find stuff. What file would you recommend implementing the keybinding class in?

Regarding API, I had done some work on this myself in parallel so I will review your work with interest. There may be a challenge to identify which API calls should be advertised for keybinding. We may want to limit to those that take no (or default) arguments. I will have a think about that. I haven't looked yet to see if you intend to retire CUIA and replace with API. They certainly overlap so maybe CUIA becomes a subset of the API.

I currently have no broadband and my mobile data has constraints so I am unable to test on my Zynthian. I will continue to code and try to hack a route back to UK to allow remote testing 😀.

riban-bw, 2019-11-08:
I have hacked into my home Zynthian so can test. Last commit implemented the keybinding class and seems to work. I am now adding documentation to the class then will add code to load configuration from file at startup.

I am not sure there is an advantage to adding multiple configurations (QWERTY, numerical keypad, etc.) I think it may prove to be an overhead that has little advantage and is irksome to maintain and it complicates the user interface (webconf). I suggest we add a default mapping for QWERTY including numerical keypad then allow the user to override that but saving to a common configuration file. (We can provide restore to defaults by hard coding those defaults in the class like I have done.) The current implementation does support multiple configurations, each in its own yaml file but I suggest we do not use that feature and only use the default file. If a use case for multiple configuration files manifests then we can extend webconf to support that.

So I will add the loading of default.yaml and then take a look at webconf.

riban-bw, 2019-11-09:
Okay - load at start-up is working but how do I get the default.yml file into zynthian/config/keybindings during an update? I added the folder and file to _feature_configurable_keybinding_ branch of zynthian-sys repository but do not see what will copy it to the correct config folder.

@jofemodo let me know the answer to above. I will start looking at webconf next.

mheidt, 2019-11-10:
I would like to have a switch to disable Keybinding completely as well.
And we need "Midi-Learn" from the webconf.
Selecting a field in webconf, pressing the key, voila.

riban-bw, 2019-11-11:
Okay - it seems to be working now if anyone wants to test. Requires these branches:

- https://github.com/zynthian/zynthian-ui/commit/cd5bd8fb530e0f31bd1dc0bfd0a298cd1c6c8928
- https://github.com/zynthian/zynthian-webconf/commit/c780c72778d05f7268c3cc0dd75ab2335b90f81f

No change to sys because the config file is optional and hard-coded defaults are used in its absence. I moved the config file to the config folder rather than a subfolder and renamed it to 'keybinding.yaml' because I don't think we need more than one configuration as explained earlier. I have added a restart of the UI after saving keybinding but this should be replaced with a simple reload when we have a mechanism of triggering events in UI from webconf.

I have not yet implemented @mheidt suggestion but will look at that next. I may change the config file structure to add an extra layer so that the enable flag can be held within the same file as the binding map. I am not sure what is meant by the MIDI-Learn feature. Do you mean that it would be advantageous to allow users to press a key (combination) in the webconf and for that to be mapped to the CUIA function? Such an implementation would be nice but may require some ingenious web coding. I have only just coded with tornado for the first time :-).

I will add the feature to disable keybinding and also to restore to default for the next iteration. When we are happy we can squash the commits and @jofemodo can consider merging to master. (I will rebase to master during the next set of changes.)

riban-bw, 2019-11-11:
I have implemented feature to disable keybinding. There is something odd about resetting to defaults which I will look at later. (Need to go and see some of Australia now - whilst I am here....) You should be able to test the implementation with:

- https://github.com/zynthian/zynthian-ui/commit/680bbb529a46e338593c73198745e40353b168b8
- https://github.com/zynthian/zynthian-webconf/commit/9fe3c2d098894515442d6f934526bcf70dbc601d

riban-bw, 2019-11-12:
Okay - I think that is it. Please review this branch. It implements the ability to configure keyboard binding, allowing each CUIA to be mapped to one or more keyboard keys with one or more modifiers and also to disable keybinding. Settings are saved to file for persistent storage.

jofemodo, 2019-11-12:
Hi @riban-bw !

I just tested a little bit. The interface is correct. We don't need more than this for this feature.

**Functional Problems:**
- When saving, for first time, the keybinding got disabled. It starts working normally after i changed something and saved again.
- Modifiers checks doesn't work correctly. Many remain unchecked,  several actions are assigned to same key combination (on webconf GUI). 

**Style Problems:**
- Table header spacing is sub-optimal
- Table itself should be centered.

**Conding tips:**
Your code is fantastic ;-)
String formatting should be done with "format". The old syntax with "%" is disouraged.

Thanks!

jofemodo, 2019-11-12:
Don't forget to pull my changes from both repos ;-)
The table looks much better now ...

riban-bw, 2019-11-12:
@jofemodo thanks for fixing the formatting. I purposely made the row height smaller than other tables to allow more overview but there is a benefit to consistent design. I meant to request opinion but it was forthcoming without the request :-) . I tried to do stuff with CSS but couldn't get it to work. I need to look at "less" templates...

I cannot reproduce either of the issues you report. Starting with no configuration file the first save does not disable key mapping for me. I set each modifier different and saved and all were preserved. Give me a detailed bug report ;-) on how to reproduce your issues.

I tried to implement scrolling table with fixed header but my failure to edit CSS meant this failed. It would be advantageous to avoid the table header from scrolling off page so users know which checkbox represents which modifier.

I wonder how best to indicate that the key cell is a comma separated list of keys. Maybe title it: "key,key,..." or add a tool-tip (not much use for touchscreen users). Maybe the fact that default settings has some multiple assignments may be sufficient for users to know / deduce this fact (and an entry in the wiki).

I don't like that a UI restart is required after making changes to key binding and it shouldn't be necessary but I don't think we have a direct connection between webconf and ui yet so implemented the restart. It is implicit but maybe it should be explicit, i.e. request user to restart UI? I can't remember how that is handled elsewhere.

jofemodo, 2019-11-13:
Of course we have direct connection with UI, my friend!!
You can send CUIA using OSC. In fact this is already done when saving MIDI options ;-)
So, simply implement a CUIA for "load_keybinding" (with a parameter) ;-)

Regarding the "Reset to defaults" ... yes, a confirmation could be fine ;-)

jofemodo, 2019-11-13:
Tomorrow i will give you a detailed bug report ... ;-)

jofemodo, 2019-11-13:
Ahhh! I just rebased the devel branches with master ... don't forget to pull!

riban-bw, 2019-11-13:
I have looked through the source but can't see where the IPC is happening when webconf saves MIDI options. Please point me at that so I can see how it is done. Meanwhile I will look at adding a confirmation dialog.

jofemodo, 2019-11-13:
Let me point you to the specific lines ;-)

https://github.com/zynthian/zynthian-webconf/blob/2933fafe5363e3e30858ee42b22566e91bacf7c3/lib/midi_config_handler.py#L460

https://github.com/zynthian/zynthian-webconf/blob/2933fafe5363e3e30858ee42b22566e91bacf7c3/lib/zynthian_config_handler.py#L112

Regards,

jofemodo, 2019-11-13:
Today i can't reproduce the problems from yesterday. Perhaps i was too tired or some of my changes solved the problems...

So ... it's merged!

Only rest to implement the CUIA for reloading the keybindings from config file.

riban-bw, 2019-11-14:
Reload keybinding via OSC is now implemented in the branch. It seems to work well.
Question: Enable / Disable Keybinding checkbox does not save configuration so the Save button needs to be pressed to assert this. This is logical (as other changes in the form are not automatically saved) but I wonder how intuitive it is. Should the checkbox save changes automatically?

jofemodo, 2019-11-14:
Thanks for the CUIA implementation. I tested and merged already (again, don't forget to pull).

I changed the "r" (record, it's ambiguous) by "a" (audio). All the "Audio Record/Play" functions are in this key, using different modifiers. I used the "alt" for "toggle", but we could change it, of course.
In the same way, all the MIDI record/play functions are in the "m".

Regarding the "enable" check, it shouldn't be needed to click save. Also, when "disabled" any button should be shown.

Regards,

riban-bw, 2019-11-14:
Enable checkbox now auto-saves.
Save button is hidden when keybinding is disabled. Reset to default button remains visible because it is still valid.
I also corrected some indenting of the html.
I reckon we are mostly there with this one!!!

jofemodo, 2019-11-15:
From my POW, we can close the task ;-)

closed 2019-11-15 by jofemodo

riban-bw, 2019-11-16:
Ticket may be closed but that doesn't stop development... In response to the suggestion to [create a cusom keypad](https://discourse.zynthian.org/t/control-your-zynthian-with-a-qwerty-or-other-keyboard/3057/42) I have added support for Num Lock and Alt Gr (right alt) modifiers to the branch. These work but the AltGr is a little tricky to configure because each key pressed results in an _interesting_ key symbol, e.g. on my UK QWERTY laptop keyboard, AltGr+'a' produces a 'á' character which is symbolised to 'ae'. AltGr+'1' gives no printable character but keysym, 'onesuperior'. This is of course a very useful feature as it allows a larger quantity of key binding and may provide users with intuitive mapping. I fear that documenting every keysym may be futile - the quantity of combinations may make such a table impractically large and may differ between languages and keyboard layouts. We may (at some point in the (far) future) consider a method of mapping more simply but for now I think the current webconf implementation is the appropriate pragmatic solution.

mheidt, 2019-11-16:
What do you mean by á is symbolised by ae? Any french around? Normally ä is symbolised by ae.

riban-bw, 2019-11-16:
The TkInter keysym for the character sent by AltGr+'a' is "ae" so to bind this key combination you need to select the AltGr modifier and enter "ae" in the key map.

mheidt, 2019-11-16:
> The TkInter keysym for the character sent by AltGr+'a' is "ae" so to bind this key combination you need to select the
> AltGr modifier and enter "ae" in the key map.
> 
Ah.My german keyboard doesn't have a alt-gr label on the a, but produces æ.We have to press the accent followed by the a
to produce the á
> [
> {
> "@context": "http://schema.org",
> "@type": "EmailMessage",
> "potentialAction": {
> "@type": "ViewAction",
> "target": "
> https://github.com/zynthian/zynthian-webconf/issues/101?email_source=notifications\u0026email_token=AA7XOQWN3JN42DXH7QDQIZ3QT6ZUVA5CNFSM4JJYM7LKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEEHNBSI#issuecomment-554619081
> ",
> "url": "
> https://github.com/zynthian/zynthian-webconf/issues/101?email_source=notifications\u0026email_token=AA7XOQWN3JN42DXH7QDQIZ3QT6ZUVA5CNFSM4JJYM7LKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEEHNBSI#issuecomment-554619081
> ",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]