zynthian/zynthian-webconf #103: WebConf Audio configuration shows "None" suffix for some field titles

riban-bw, 2019-11-09:
Many field titles in the Audio (and possibly other) webconf pages are suffixed with "None".

**To Reproduce**
1. Navigate to Hardware->Kit and select, "Custom"
2. Navigate to Hardware->Audio

**Expected behaviour**
Field titles should show correctly, e.g. "Soundcard"

**Actual behaviour**
Field titles should show incorrectly, e.g. "SoundcardNone"

**Setup:**
HARDWARE
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+ light
Display: Generic HDMI Display
Wiring: I2C_HWC
GPIO Expander: No detected
SYSTEM
OS: Raspbian GNU/Linux 10 (buster)

SOFTWARE
zyncoder: master (8b7bfbd)
zynthian-ui: feature_configurable_keybinding (19788a1)
zynthian-sys: feature_configurable_keybinding (a4fd281)
zynthian-data: master (075b4f7)
zynthian-webconf: master (f7c5327)

jofemodo, 2019-11-09:
It's a regression. I fixed this error 1-2 weeks ago ...

jofemodo, 2019-11-09:
OK! I only fixed the problem in one controller and the problem affect several others. 5 mins ;-)

commit a4b0de13 (2019-11-09): Fix #103: Audio & Display configuration shows "None" suffix for some field titles
closed 2019-11-09 by jofemodo