zynthian/zynthian-webconf #108: Ability to create bug report with system data pre-populated

riban-bw, 2019-11-29:
When creating a bug report one is asked for details of the Zynthian system, e.g. SD Image, Hardware, Software. It can be irksome and error prone to manually enter this detail for each bug report. It would be advantageous to be able to populate a bug report with this information automatically.

**Describe the solution you'd like**
A method to create a bug report within the issue tracking system with the common system data pre-populated. This could from a button in webconf which could direct to issue tracker, passing appropriate data which is mostly already available within webconf overview.

**Describe alternatives you've considered**
Issue tracker could have templates or preset data for each user but this is error prone as user must add the data manually, maintain it and may be reporting an issue for a system with a different configuration. Also issue tracker may not offer this feature.

**Additional context**
[image]

mheidt, 2019-11-29:
The reason for the former being on hold was that I need a GitHub user I can use to add new issues. During development it should not send emails when I submit new isses

riban-bw, 2021-03-19:
I have submitted PR https://github.com/zynthian/zynthian-webconf/pull/158 to implement this feature.

riban-bw, 2021-03-19:
I reported issue zynthian/zynthian-issue-tracking/issues/402 using this method and it seemed to work okay.

riban-bw, 2024-03-12:
OMG! I did this again and made the exact same mistake!!! With age comes wisdom but also amnesia. Anyway, a work-around for the issue was done by @jofemodo by making this a button only available in the main dashboard view. It is implemented in oram so I will close this ticket.

closed 2024-03-12 by riban-bw