zynthian/zynthian-webconf #110: Change MIDI channel base to 1 (1-16)

jofemodo, 2019-12-03:
All MIDI channel references in webconf should be 1-based:

+ MIDI rule builder
+ MIDI log

mheidt, 2019-12-03:
Midi Log is 1-based.
When the Rules are changed, it needs changes in the rule builder as well

mheidt, 2019-12-18:
> Assigned  #110 to  @mheidt .

As of now into master?

jofemodo, 2019-12-19:
It's not implemented in the router yet. I only assigned the task ...
I will tell you when it's ready ...

mheidt, 2020-01-21:
<children in the car>
Are we there yet?

jofemodo, 2020-01-21:
< daddy driving the car >
Nop

riban-bw, 2021-07-11:
What about now?

jofemodo, 2021-07-11:
Let me refresh the task on my mind ...

riban-bw, 2024-03-12:
I think this is now working as desired. I will close ticket. You can open a new ticket in the issue tracking system if required. (We have arrived at our destination and it isn't Disneyland!)

closed 2024-03-12 by riban-bw