zynthian/zynthian-webconf #112: New Feature: Mixer Configurator

jofemodo, 2019-12-09:
We need a mixer configurator that allows to choose the "mixer controls" we want to have in the UI's mixer panel:

+ A "tag selector" widget may be used for the view part

+ The mixer engine already implements the zynapi methods that allow to obtain a list of available "zcontrollers". We should implement webconf zcontroller-widgets for controlling the mixer from the webconf side. This zcontroller-widgets could be reused for implementing layer/engine control in the future.

jofemodo, 2019-12-09:
Recommended widgets:

+ https://select2.org
+ https://selectize.github.io/selectize.js/

jofemodo, 2019-12-12:
Hi @mheidt !
What do you think about this task? Would you like to take it? ;-)

mheidt, 2019-12-12:
I haven't actually checked out the mixer branch yet. Wanted to do it yesterday.
You haven't been online yesterday much, so that I though, you would be working on this and surprise us again.
Saying, yes I could help the next days. But I don't have a picture of the demand yet.

jofemodo, 2019-12-12:
OK! Fantastic! Of course, first you should take a look to the new Mixer panel ...

Regarding these last days ... i', quite busy with the new "transport" engine and the BPM control of MIDI player. We are going step-by-step into the micro-DAW road way ;-)

mheidt, 2019-12-12:
Ok, if you ask for micro-DAW:
https://github.com/zynthian/zynthian-ui/issues/182

mheidt, 2019-12-12:
The new mixer panel is the Volume knob in Audio Player? Or did I miss anything?

mheidt, 2019-12-12:
I created a mixer branch and implemented what I understood so far...which is not much :)

jofemodo, 2019-12-15:
The mixer panel have to be shown as "MIXER" in the layer list. It's not related with the Audio Recorder volume control, that is a completely different feature.
Probably you have to click "save" in the weconf hardware->audio settings before the MIXER panel works correctly.

mheidt, 2019-12-19:
I updated everything and ensured, that all repos are master.
I saved the audio setting.
But I don't see the MIXER panel in the layers view. 
Could you post a screenshot?

jofemodo, 2019-12-19:
https://discourse.zynthian.org/t/new-features-mixer-panel-midi-player-bpm-control/3560/20

mheidt, 2019-12-21:
What exactly do you have in mind for webconf?
Instead of the existing volume slider you want a different popup screen?
Showing all available sliders with a checkbox. All selected checkboxes will be stored in the text field of the main screen.
And sliding the slider changes the volume (et.al) in realtime?

jofemodo, 2019-12-28:
More or less ... but not all the controls are sliders. Some of them are "selectors" and some "booleans". The mixer engine implements the method "zynapi_get_controllers" that returns an array with the available mixer controllers as zynthian_controller objects.

mheidt, 2020-01-10:
@jofemodo,
I have a websocket connection now and I am about to change audio_mixer_handler.
There is this function now
	def do_update_audio_mixer(self, zcontroller_name, value):
		#zynthian_engine_mixer.init_zynapi_instance()
		#zynthian_engine_mixer.ctrl_list = []
		logging.debug('Changing value of {} to {}'.format(zcontroller_name, value))

What do I have to do to change the value through the api?

logging output: Changing value of Playback_DAC to 52

mheidt, 2020-01-11:
Next stop callback...I want the webconf change the values, when you turn the rotaries

mheidt, 2020-01-11:
When you take the branch mixer and zynthian-ui/mixer2, you have something to test.
I am thinking of letting zynthian_engine_mixer establish a websocket connection and sending a message each time _set_value is called.
What do you think, @jofemodo?

mheidt, 2020-01-12:
and i think, that the zynthian_engine_mixer is not changing the value in the ui, if set_value is called externally (by webconf)

jofemodo, 2020-01-13:
That's is the expected behaviour. We are calling the zynapi functions as a "library", creating a separated instance of the UI objects, no accessing the running UI objects. To do so, we should create a RPC (OSC or HTTP) call interface. It's not complex, as the more difficult part is already done, but then, the webconf won't work if the UI is not running. I would do it for the set_value only ...

mheidt, 2020-01-13:
we need a shared memory or interface between zynthian-ui and webconf. in order to communicate changes in both directions.
webconf->zynthian-ui we could force the OSC call?
But other way around is tricky. As I suggested a websocket connection zynthian-ui as client to webconf?
Is there another shared memory option in python?
Should we merge the existing code before adding this features?

jofemodo, 2020-01-13:
I think HTTP is the way ;-)

mheidt, 2020-01-13:
You mean, that I should include the audio mixer rest interface again (deleted it in my branch).
the engine is calling that rest interface, when a value is changed.
And the ui opens it's own rest interface, that the webconf is going to use?

mheidt, 2020-01-13:
Now there is a Rest call you can change the value in the webconf with.

I tested it with curl
curl -v --cookie "user=2|1:0|10:1578094718|4:user|8:cm9vdA==|aa5f294d2b43ebf3c192e5bf1432d9b49a220560e6c357f09b3671c34f9c3c29" -X POST http://localhost:8089/hw-audio-mixer/Playback_Master/20

The name of the controller needs underscores instead of blanks. (might be the same issue @jypma talked about...which should be fixed with the mixer/mixer2 branches)

@jofemodo  
Could you integrate it into the engine?
The authentication (user cookie) of the Rest call might be a challenge.

And when you add a Rest interface into the engine, I could call it from webconf to change the value, when the webconf is changing the value

jofemodo, 2020-01-14:
I think it's time to merge with master. The current functionality is enough for most of use-cases and the tool in its current status is works pretty fine.

The 2-way UI-webconf synchronization is not trivial and, IMHO, has low priority, so it can wait a little bit. I would like to  put some mortar & bricks before ... 

I just commited a few optimizations. Please, pull my changes and compact all your uncommented changes in a nice way, so i can merge with master ASAP.

Thanks a lot!

jofemodo, 2020-01-15:
OK! It's done. I had a nice love/hate history with "git rebase" ... and finally love was the winner!!
Anyway, please ... never more!! ;-DD

BTW, it's merged!

Thanks!

closed 2020-01-15 by jofemodo