zynthian/zynthian-webconf #115: Add Convert Snapshot to Preset in WebUI

Baggypants, 2020-01-05:
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Zynthian stores its configuration internally in it's own snapshot format. It would be helpful to be able to extract that configuration information as pre-sets for the various engines.

**Describe the solution you'd like**
The WebUI should have a button or functionality within the presets page that can extract the information from a zynthian snapshot or subsnapshot and save it as a preset, optionally download it to using the users local browser or even offer to upload it to https://musical-artifacts.com using the users pre-created account. For the direct engines like ZynAddSubFX, SetBFree and Aeolus this would be in the engine format, for LV2 plugins this would be in the LV2 preset format.

**Describe alternatives you've considered**
Using SSH XForwarding, editing parameters in there and using the engine save option where available.

jofemodo, 2020-01-05:
This is a good idea. I will implement it ASAP ;-)

Baggypants, 2020-01-06:
Or, "save as preset" from zyn-ui might make more sense as snapshots can have multiple layers and effects plugins associated.

mheidt, 2020-01-24:
I prepared the following:
[image]

It's in the gh115-export-preset branch.

@jofemodo, your turn?

jofemodo, 2020-01-24:
Ohhhh! Great!

It's my turn, then ;-)

mheidt, 2020-03-04:
Hm, the merge with gh126 will be tricky.
Tell me, when you have something. I will rebase then.

jofemodo, 2021-07-05:
As we are going to implement "Preset Saving" on UI, i think this is not needed any more.
If somebody needs to generate a preset from a saved snapshot, he could load the snapshot and save the preset from the UI.

Thanks

closed 2021-07-05 by jofemodo